Ultimaker/Cura #17698: slicing failure after upgrading from Cura 5.5 to 5.6

oliof, 2023-12-18:
### Cura Version

5.6.0

### Operating System

POP!_OS Linux 20.04

### Printer

M-Minion (custom FFF profile that worked in Cura 5.5)

### Name abnormal settings

Model also fails slicing with the default draft profile

### Describe model location

rotating the model also yields a slicing failed issue

### Describe your model

Model sliced successfully in Cura 5.5

### Add your .zip here ⬇️

File includes the 3mf file, the STL, and the FreeCAD sources

[slicingfailed.zip](https://github.com/Ultimaker/Cura/files/13708229/slicingfailed.zip)

5axes, 2023-12-19:
Strange no issue for me. 
[image]

oliof, 2023-12-27:
after removing a bunch of cache directories (`~/.cache/cura` and `~/.cache/Ultimaker B.V.`) as well as moving both config directories out of the way and back to their original place (`~/.config/cura` and `~/local/share/cura/`), I can slice again with Cura 5.6. I assume some kind of cache corruption occurred.

oliof, 2024-01-21:
Running into slicing issues again, looks like rounding / close-to-zero issues. In the attached 3mf, slicing fails if wall width is set to 2.5, but succeeds when set to 2.51. Likewise, slicing fails if line width is set to 0.5, but succeeds when set to 0.49

[project.zip](https://github.com/Ultimaker/Cura/files/14001068/project.zip)
[cura_log.zip](https://github.com/Ultimaker/Cura/files/14001073/cura_log.zip)

oliof, 2024-01-21:
Previous project had bottom layers set to 999999 for some reason. Here is another similar profile with less weird settings. Slicing fails when I enable tree supports, and succeeds as soon as I set branch diameter to 4.99 or 5.01 instead of 5.0

[Handtuchhalterhalter2-Body-new.zip](https://github.com/Ultimaker/Cura/files/14001128/Handtuchhalterhalter2-Body-new.zip)
[cura_log_new.zip](https://github.com/Ultimaker/Cura/files/14001143/cura_log_new.zip)

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well?
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic

oliof, 2024-04-03:
I already saw some improvements in 5.7beta1 but did not try this particular model yet. Will do soon and report back.

oliof, 2024-04-03:
Both models mentioned above slice without issue in Cura 5.7.0. Thanks for improving Cura with every release!