Ultimaker/Cura #17701: Cura 5 Unsauberes Slicen

schlossi82, 2023-12-19:
### Cura Version

5.5.0

### Operating System

Windows 10

### Printer

JGaurora A5S

### Describe your model

In Cura V4 is perfekt. All newer Versions are Problem

### Add your .zip here ⬇️

[Cura ab V5 JA5A5S_Slug für 8mm Kugeln V2 3 - UltiMaker Cura.zip](https://github.com/Ultimaker/Cura/files/13712603/Cura.ab.V5.JA5A5S_Slug.fur.8mm.Kugeln.V2.3.-.UltiMaker.Cura.zip)

All Cura Versions 5.... I get always shit slicing for that. Itested every Version. Cleaned installations. Only Cura V4 make a clean slicing.
I tested on 3 different PC´s always the  same Problem.

5axes, 2023-12-19:
We need an UltiMaker Cura project file (**3MF**) for anyone to be able to help you with this.
guide: https://support.ultimaker.com/s/article/1667411030205

5axes, 2023-12-19:
Well Satisfy by the result. 

[image]

5axes, 2023-12-19:
Could be also interresting to test with a very small cutting resolution . On this very small Model

0.005 mm
[image]

schlossi82, 2023-12-23:
Hello, yes look's very good. So I have it with the old Cura v4. All v5 is the z position some where and it is not possible to slice to the end of line. V4 no problem with the same profile. Sorry for the bad English.

5axes, 2023-12-23:
Of course if you try to have the same result between V5 and V4 with the same profile you will be not succesfull. It's two different slicing engine, so you must change some settings if you want to have a correct result.