Ultimaker/Cura #17703: Issue #17702: re

oscargg, 2023-12-19:
### Cura Version

5.6 & 5.4

### Operating System

w10

### Printer

anycubic chiron from library or diy setting

### Describe your model

slicing error 

### Add your .zip here ⬇️

[bass drum claw.zip](https://github.com/Ultimaker/Cura/files/13714262/bass.drum.claw.zip)
i've added a screenshot of error msg, and a material setting profile that creates the error
if you can't recreate the issue, i will try to deinstall all cura version (actually 15.04, 5.3, 5.4, 5.6 installed) and reinstall only last ver.
tks

5axes, 2023-12-19:
Sorry No issue : Windows 10 like you  Cura 5.6.0 . But As every release are independant  I don't think it's a good idea to unstall all cura release. 

[image]

5axes, 2023-12-19:
@GregValiant Do you have any issue on your side ?

closed 2024-04-03 by HellAholic