Ultimaker/Cura #17707: Support for the Elegoo Neptune 4 Pro, please.

Lance61, 2023-12-19:
### Is your feature request related to a problem?

I buy a new product that relies upon a 3rd-party product that doesn't support it.

### Describe the solution you'd like

Support for the Elegoo Neptune 4 Pro, please.

### Describe alternatives you've considered

n/a

### Affected users and/or printers

Elegoo Neptune 4 Pro

### Additional information & file uploads

n/a

udihq, 2023-12-20:
[image]

What do you mean? They N4P **IS** supported.

vadim-danilchik, 2023-12-20:
It is missed on 5.6.0 installation
[image]

[image]

RyanWhipple, 2023-12-25:
#16819 adds support and has been merged.  Hopefully this is included in the next release, but in the mean time you can copy the new/updated files into your Cura installation and they will be available for your use.  That's what I'm doing until these changes make it into a release.

#17576 is still open and adds & updates some additional configurations if you are interested in using those resources.  I haven't used these yet, but it looks like a number of people have tested them and found them to work.

Hope this helps!