Ultimaker/Cura #17713: There is the way to completely eliminate the visible seam.

vgdh, 2023-12-20:
### Is your feature request related to a problem?

# Hide the Seam Completely
## Explanation
Every closed loop will eventually have a visible seam. However, if there is no specific point where the line ends, there is no seam, right?
1. Begin the line with a height of 0 and gradually increase the height of the nozzle and the line over some length until it reaches the full height of the line.
2. Move through the loop until reaching the end of the line, but continue extruding the filament due to the initial zero line height. Systematically reduce the line's height while maintaining the same nozzle height until reaching the end of the starting slope.
3. Move on to the next perimeter or infill.

That's it. The algorithm is remarkably simple, though the result is astonishingly good. 
It is not an easy task to post-process already existing GCODE files, but if it works at the slicer level, it will work very well.

## The python code
The repository includes an [example](https://github.com/vgdh/seam-hiding-whitepaper/blob/main/main.py) demonstrating a basic implementation of the algorithm for printing a ring. This implementation allows specifying parameters such as the number of lines in the ring, layer height, line width, speed, etc.

Also there is an [example](https://github.com/vgdh/seam-hiding-whitepaper/blob/main/nozzle%200.4%20line%200.3x0.8.gcode) of gcode file for 0.4 nozzle line 0.3x0.6 layer height 0.3

## Photos of the Seam Details on the Printed Example.
The first layer (the top on the first photo) was printed without the algorithm to get better adhesion.
[image]
[image]
[image]

In the photo below you can see there is no gap between the lines.
[image]
[image]
[image]
[image]
[image]

### Describe the solution you'd like

I created and tested the algorithm to hide the seam completely. It would be nice if it will work within the slicer.

### Describe alternatives you've considered

_

### Affected users and/or printers

all the community

5axes, 2023-12-20:
Must be tested. 
[image]

5axes, 2023-12-20:
Edit : Not totaly convice .. on my Ender-3 I still have a seam , in fact 2 seams . One at the beginning the other one at the end of the  start area . And I'm not sure that a printer can respect the instruction of Z=0.43124999999999997

vgdh, 2023-12-20:
works well on the cheepest kingroon kp3s.

5axes, 2023-12-20:
I think that for some machines, this could be a solution. 

On a bowden Extruder doesn't get a nice result :

**External Start of slope Area**
[image]

**External End of slope Area**
[image]

Gcode tested on Ender-3 : https://github.com/5axes/seam-hiding-whitepaper/blob/main/ender-3%20nozzle%200.4%20line%200.25x0.4.gcode

nallath, 2023-12-20:
Unfortunately, this looks suspiciously like a method that has been patented. If i'm right (and it's indeed patented), this will prevent us from adding it :(

fieldOfView, 2023-12-20:
What if someone were to implement this as a CuraEngine plugin though? Ultimaker cannot help what external contributors implement, right?

vgdh, 2023-12-20:
> What if someone were to implement this as a CuraEngine plugin though? Ultimaker cannot help what external contributors implement, right?

that is true

5axes, 2023-12-20:
> What if someone were to implement this as a CuraEngine plugin though? Ultimaker cannot help what external contributors implement, right?

If you don't want to put it on the market place, why not, but personally Ultimaker refused one of my plugins because it used a common font that was not free to reuse to make an STL model, for example. So I think it would be the same here.

nallath, 2023-12-21:
The patent I was worried about is this one https://patents.google.com/patent/US8349239B2/en but it only makes claims about the XY plane. So by including the Z plane it should put you in the clear (but I am not a lawyer, so do not take my word for it)

nallath, 2023-12-21:
> > I think that for some machines, this could be a solution.
> > On my bowden Extruder doesn't get a nice result
> 
> I think bowden can't provide the right accuracy of flow rate. May be it for direct machines only.

I think that should be possible. You can't change the flow quickly, but you can change the speed quickly. So instead of changing the flow, you just change the speed of the nozzle. That's what we do for Arachne currently

5axes, 2023-12-21:
> The patent I was worried about is this one https://patents.google.com/patent/US8349239B2/en but it only makes claims about the XY plane. So by including the Z plane it should put you in the clear (but I am not a lawyer, so do not take my word for it)

It's not the same solution and I don't think this point could be cover by this patent .. There are an other Enhancement request on this Github concerning this patent and how to reduce Seam  : https://github.com/Ultimaker/Cura/issues/5728#issuecomment-1604470163  This one is exactly what the patent describe.

vgdh, 2023-12-26:
I have written some post-processing [script](https://github.com/vgdh/seam-slope-postprocessor/blob/main/postprocessor_seam_slope.py). Feel free to test, modify, or contribute.

vgdh, 2024-01-01:
Now the script processes only the outer perimeters, printing the inner ones as usual.

vgdh, 2024-01-01:
bender likes it very much 😁
[image]
[image]

GregValiant, 2024-01-02:
[image]

vgdh, 2024-01-31:
This feature in now implementing by OrcaSlicer: https://github.com/SoftFever/OrcaSlicer/pull/3839
Feel free to participate.

5axes, 2024-01-31:
Looks like I'll have to do some more OrcaSlicer this WeekEnd ;)

Sophist-UK, 2024-03-02:
Also now implemented implemented in [Prusa](https://github.com/prusa3d/PrusaSlicer/issues/11621) and [I have a different OrcaSlicer reference](https://github.com/SoftFever/OrcaSlicer/issues/3211) slicers.

This picture is from the [Prusa PR](https://github.com/prusa3d/PrusaSlicer/issues/11621):

![Prusa example from their PR](https://private-user-images.githubusercontent.com/141463653/281572451-a0da4fbc-8012-4e46-bac7-a0920e24cd36.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkzODMzODIsIm5iZiI6MTcwOTM4MzA4MiwicGF0aCI6Ii8xNDE0NjM2NTMvMjgxNTcyNDUxLWEwZGE0ZmJjLTgwMTItNGU0Ni1iYWM3LWEwOTIwZTI0Y2QzNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMwMlQxMjM4MDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04NzQzZDM1Zjk0OGFkYWI5MWYzOTg4YzdhNTRlNmQ2OTVjNTYyYTA4YTBlYjMwOTlkZjNhYjExN2IzY2M5NDFjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.5CMU4eF4qRphDJWiU3qwAjEQoF36cX02RqSxh5Bakts)

See also https://www.youtube.com/watch?v=vl0FT339jfc

As an interim could anyone implement this as a PostProcessing script?

5axes, 2024-03-02:
For the moment, this function is only integrated into Orca : https://github.com/SoftFever/OrcaSlicer/pull/3839

There has been a lot of work between the original script and the version that has just been integrated. So I'm not sure that integration via a script in Cura is possible. The Cura team will have to integrate this function in Cura natively, but I'm not sure that this is one of their priorities.  There are still steps in development, notably with an adaptive option where the **scarf joint seam** would only be integrated on contours without sharp angles, as in this case the use of this option is counter-productive.

5axes, 2024-03-03:
New option in OrcaSlicer : https://github.com/SoftFever/OrcaSlicer/pull/4317

printingotb, 2024-03-05:
This is somewhat of a gamechanger. Would love to this in Cura :D

GregValiant, 2024-03-06:
@5axes Sorry about this late reply.  I just saw the updated comments.
I changed the retractions in that gcode file to 6.5.  My results were exactly the same as yours with two seams on the exterior about 180° apart.  There were also two seams on the interior.
This was sliced with Orca 2.0dev.  The seam on the backside was unobtrusive but visible.  This scarfed seam wasn't too good.  It may have been better if there was some extra prime, but that is something I've never had to use.
[image]

As with the example gcode - there was a lot of cross-print travel which lead to stringing (PETG).  That may have been a result of my inexperience with Orca.
Between the stringing, under-extrusion, and poor seam quality my opinion (as of 8:01AM EST) is "Not yet".
That is of course subject to change on a whim.

5axes, 2024-03-06:
You can do better ... maybe it's time to leave Cura and invest time in another solution :)

GregValiant, 2024-03-06:
I like the customization possible with Cura.  I also prefer the interface (as clunky as it is).  I prefer the supports as well.
The "I refuse to slice this" bug hasn't affected me much.  It mostly occurs with organic shapes and I don't print very many of those.  Industrial type things haven't been a problem.

I have ported all 28 of my post-processors to VB.  I can open a gcode file created by Prusa, Orca, Bambu, and IdeaMaker, as well as Cura and Creality - post-process them, and write a new gcode file.  The VB code needs to be cleaned up and organized better, but they work.

moccor, 2024-03-06:
Among many other reasons, Cura has cloud backups and plugins like "Filament Settings" that make it have far better customization and saves lots of time compared to others.

I have a Ender V3 SE with a pretty well-tuned Klipper atm, is there some easy way to try this scar method to provide some direct-drive (0.6mm retraction). I assume I just download the gcode file and edit it as needed?

GregValiant, 2024-03-06:
The Gcode is sliced in Relative Extrusion.  You can go through and replace the retract and prime lines with the retract distance of your choice.  "As received" the retract distance is 0.5.
I changed the retract distance to 6.5 and I copied the startup and ending gcodes from another file and pasted them into the scarf gcode.  It worked fine.

nerdguy1338, 2024-03-16:
> Unfortunately, this looks suspiciously like a method that has been patented. If i'm right (and it's indeed patented), this will prevent us from adding it :(

Isn't this too simple to be patentable?

5axes, 2024-03-16:
You can have a patent on a "simple" Idea. But concerning this discussion and the mentionned patent, it's not the same solution so in the present case there is no limitation to use it in Cura. 
Second pointthe scarf seam is not any more a simple solution. After a lot of tests and prototype the solution and number of parameters is now much more significant

T9Air, 2024-04-09:
What is the difference between this and vase mode?

vgdh, 2024-04-09:
> What is the difference between this and vase mode?

It's already been implemented in OrcaSlicer feel free to try.
wase does only one line width wall and this method can do any number of lines.

Sophist-UK, 2024-04-09:
Base mode does a single **continuous** wall spiralling round and round.

Scarf joins are like a wooden scarf joint. It starts a wall (any wall) by creating a gradually thickening extrusion (raising the height and extending more) over a short distance until it gets to the normal layer thickness, creating a sloping start to the layer. Then it does the rest of the wall normally. When it gets back to the start point it goes over the sloping part again, gradually reducing the extrusion amount so that it fills the sloping area to the normal height. 

I have tried this with orca and it is NOT a magic solution to seam bobs caused by oozing - you need to get your retractions and/or linear advance right first.

Sophist-UK, 2024-04-09:
If you think about how this works, the total amount extruded is (in theory at least) identical to a normal wall, so I think it should easily be within the bounds of possibility to achieve this with a post-processing script.

5axes, 2024-09-06:
https://github.com/Ultimaker/Cura/pull/19571

vgdh, 2024-11-06:
This feature has been implemented in the latest cura [release](https://github.com/Ultimaker/Cura/releases/tag/5.9.0-beta.2).