Ultimaker/Cura #17720: Setting Slicing Tolerance to Exclusive disables Build Plate Adhesion

rbgregory, 2023-12-21:
### Cura Version

5.6.0

### Operating System

Mac OS Sonoma 14.2

### Printer

Ender3 Pro

### Reproduction steps

1. Enable Build Plate Adhesion, doesn't matter which, I selected Skirt.
2. Under Experimental, set Slicing Tolerance to Exclusive.
3. Slice and check Preview.  No skirt (or whatever you selected) is generated.
4. Select any other Slicing Tolerance (Middle or Inclusive) and repeat.  Skirt is generated.

### Actual results

When Slicing Tolerance is set to Exclusive, Build Plate Adhesion is disabled even if specified.  When Slicing Tolerance is set to Middle or Inclusive, Build Plate Adhesion will function normally.

### Expected results

Build Plate Adhesion should not be affected by Slicing Tolerance.

### Add your .zip and screenshots here ⬇️

[image]
[image]

5axes, 2023-12-21:
We need an UltiMaker Cura project file for anyone to be able to help you on this issue.
guide: https://support.ultimaker.com/s/article/1667411030205

In my case Setting Slicing Tolerance to Exclusive doesn't have any effect on the Build Plate Adhesion. So it could be the combinaison of several settings... Or just a specific case due to your model.

[image]

rbgregory, 2023-12-21:
[qb20_rudder_3.zip](https://github.com/Ultimaker/Cura/files/13744295/qb20_rudder_3.zip)

rbgregory, 2023-12-21:
Attached 3mf (zipped).  Model is at (0,0,0) so I don't think that's it.

5axes, 2023-12-21:
Could be also due to the Model Geometry error.

[image]

Note **Print Thin Walls** in this case solved the issue :

[image]

rbgregory, 2023-12-21:
Looks good, works for me.