Ultimaker/Cura #17725: Slicing Failed

bfrensley, 2023-12-21:
### Cura Version

5.6.0

### Operating System

MacOS Catalina

### Printer

Anycubic Vyper

### Describe your model

MeshFix was performed and still failed.

### Add your .zip here ⬇️

[AV_DT - Wild West - Water Tower - Top.3mf.zip](https://github.com/Ultimaker/Cura/files/13744858/AV_DT.-.Wild.West.-.Water.Tower.-.Top.3mf.zip)

5axes, 2023-12-21:
An other "**Print Thin Walls**" issue : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

[image]

Duplicate Slice Failed due to **Print Thin Walls**

https://github.com/Ultimaker/Cura/issues/17725
https://github.com/Ultimaker/Cura/issues/17698
https://github.com/Ultimaker/Cura/issues/17679
https://github.com/Ultimaker/Cura/issues/17676
https://github.com/Ultimaker/Cura/issues/17671
https://github.com/Ultimaker/Cura/issues/17606
https://github.com/Ultimaker/Cura/issues/17560
https://github.com/Ultimaker/Cura/issues/17528
https://github.com/Ultimaker/Cura/issues/17503
https://github.com/Ultimaker/Cura/issues/17499
https://github.com/Ultimaker/Cura/issues/17470
https://github.com/Ultimaker/Cura/issues/17467
https://github.com/Ultimaker/Cura/issues/17457
https://github.com/Ultimaker/Cura/issues/17456
https://github.com/Ultimaker/Cura/issues/17496
https://github.com/Ultimaker/Cura/issues/17478
https://github.com/Ultimaker/Cura/issues/17439
https://github.com/Ultimaker/Cura/issues/17438
https://github.com/Ultimaker/Cura/issues/17410
https://github.com/Ultimaker/Cura/issues/17367
https://github.com/Ultimaker/Cura/issues/17344
https://github.com/Ultimaker/Cura/issues/17177
https://github.com/Ultimaker/Cura/issues/16688
https://github.com/Ultimaker/Cura/issues/16515
https://github.com/Ultimaker/Cura/issues/16417

MariMakes, 2024-01-18:
Hi @rbenites,

Quick update from our side 👋
We are introducing a new tool to analyze slicing crashes like the one you reported. 💪

Even with the changes we made in the latest versions of Cura we still see that your model crashes. 🫤 But we have added your project file to our analyzing tool to start finding crashes that are similar to yours, which helps us to narrow down what the cause is of your crash so we can resolve this for you and others.

Hopefully, we can find a solution soon 🤞

closed 2024-04-03 by HellAholic