Ultimaker/Cura #17726: [5.6.0] significantly slower slicing compared to 5.3.1

Block137, 2023-12-21:
### Cura Version

5.6

### Operating System

Windows 11

### Printer

Custom Toolchanger, Klipper

### Reproduction steps

1. Install Cura 5.6.0 (machines and profiles from 5.3.1 were automatically available, new parameters left at default)
2. Disable Optimize Wall Printing Order to rule out issue 17104
3. Slice a large model (combination of both physical size and polygon count)
4. Compare slicing time with 5.3.1

### Actual results

5.6.0 (as well as 5.5) slicing time is significantly longer than 5.3.1
One of my model took less than a minute in 5.3.1 compared to 37 minutes in 5.6.0 (500MB stl, 584g, 3 days print)

### Expected results

Comparable or faster slicing time between versions unless the newly added parameters have been altered.

### Add your .zip and screenshots here ⬇️

Benchy comparison [3mf & log.zip](https://github.com/Ultimaker/Cura/files/13744795/3mf.log.zip)
*This video is sped up x16
https://github.com/Ultimaker/Cura/assets/23495338/c56e29f2-a106-498e-b0bd-347611f00a17

Log for another large and detailed model. 
Line 25445 says the slice took 37 minutes. The same model took less than a minute in 5.3.1.
(this is paid model so I can't provide a project file) [cura.log](https://github.com/Ultimaker/Cura/files/13744819/cura.log)

5axes, 2023-12-21:
https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7791408

Vandresc, 2024-02-02:
For our internal reference CURA-11129 , the issue was addressed for the next release. Thank you for your feedback!

closed 2024-02-02 by Vandresc