Ultimaker/Cura #17731: Creality Ender3 V3 KE

Ebbsfleet1968, 2023-12-22:
### Is your feature request related to a problem?

Just purchased a Creality Ender3 V3 KE which was released in November 2023, but the printer is not found on my network by the Cura software and is not located using the printers IP address (although other devices and software locate the printer on the network).  As a new printer the Ender3 V3 KE does not feature on the drop down menu for Creality printers in Cura.

### Describe the solution you'd like

 I would like to request the Ender3 V3 KE is added as soon as possible to the Cura drop down list of Creality printers and this may solve the problem

### Describe alternatives you've considered

If I can't use LAN printing using Cura I'll have to use an alternative slicer software package

### Affected users and/or printers

Anyone with a Creality Ender3 V3 KE who want to use the Cura software for slicing

5axes, 2023-12-22:
You can use  from creality  : 

Creality slicer, an Ultimaker Cura clone of a Release 4.8 of Cura  : Creality_Slicer-4.8.2-build-254-win64
or
Creality Print the new software of Creality : V4.3.8.6984-win64-Release  seems to be based on a 5.3 Christmas tree version 

In these slicers the Creality Ender3 V3 KE is included

vadim-danilchik, 2023-12-24:
Pull request is already ready, I hope it will be merged into the next version
https://github.com/Ultimaker/Cura/pull/17519

Ebbsfleet1968, 2023-12-24:
Thanks Vadim - Merry Christmas!