Ultimaker/Cura #17737: Model failed to slice correctly

dwightkelly, 2023-12-22:
### Cura Version

5.5.0

### Operating System

MacOS Monterey

### Printer

Ultimaker S5

### Reproduction steps

Selected standard setting "Fast Engineering 0.2mm" with no supports & adhesion enabled. Infill was 40% triangles. Part imported, rotated 90 degrees and centered on build plate. It was sliced with an time estimate of 4 days 2h3m. 

### Actual results

When I previewed slicing results only the top layer was printed. The rest the model was shown as gray. I then checked the model for manifold, overlaps, face orientation, etc and found no issues.

### Expected results

Model sliced correctly for entire height of model.

### Add your .zip and screenshots here ⬇️

[image]
[UMS5_Curved_Sides.3mf.zip](https://github.com/Ultimaker/Cura/files/13754057/UMS5_Curved_Sides.3mf.zip)

5axes, 2023-12-22:
Wouah... really need some basic training.
[image]

dwightkelly, 2023-12-22:
@5axes not sure what your comment is meant to imply but I've very experienced and have 6 3d printers including commercial machines. I tried numerous settings and never got it to slice correctly.

5axes, 2023-12-23:
> I've very experienced and have 6 3d printers including commercial machines. I tried numerous settings and never got it to slice correctly.

That's exactly why I wrote "Wouah", which expresses my astonishment. It's time to take a break for the holidays. There's nothing in this issue that isn't normal Cura behavior. When we simulate movements, the current layer is in color and the layers underneath are displayed in gray. I'm sorry, but I don't see what your problem is. 

[image]

dwightkelly, 2023-12-23:
@5axes  I’ve looked at your responses to issues in this project and they generally aren’t helpful. The tone is very condescending. If you’re gonna take the time to review the issue and respond, why don’t you just help the person?

For this issue it was my fault as I jump between slicers and forgot that cura has a depth slider on right and the replay controls on bottom.  This issue can be closed.

closed 2023-12-23 by dwightkelly