Ultimaker/Cura #17739: Cura fails with every model

hd42, 2023-12-22:
### Cura Version

5.6.0

### Operating System

Ubuntu Linux

### Printer

Elegoo Neptune 4 plus

### Describe model location

default

### Describe your model

simple pyramid

### Add your .zip here ⬇️

[EN3P_Pyramide.zip](https://github.com/Ultimaker/Cura/files/13754513/EN3P_Pyramide.zip)

5axes, 2023-12-22:
Sory no issue for me : 
[image]

but in your project the printer is detected as **Anycubic Kobra Youtube** ????
[image]

hd42, 2024-01-04:
[CFFFP_TOP2.zip](https://github.com/Ultimaker/Cura/files/13829282/CFFFP_TOP2.zip)
I tried creating a new printer profile, but still get the same error.

5axes, 2024-01-04:
Still no problem for me :  
[image]

hd42, 2024-01-04:
I disabled all plugins and now it works. I'll try to enable them one after the other to find out which one caused the problem.

5axes, 2024-01-04:
similar to https://github.com/Ultimaker/Cura/issues/17551 ?

hd42, 2024-01-05:
The culprit appears to be the CuraEngineTiledInfill 0.2.0 plugin.

hd42, 2024-01-05:
Thank you for your help, @5axes.

closed 2024-01-05 by hd42