Ultimaker/Cura #17741: Tree supports causing slicing issue

amazingoomoo, 2023-12-22:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Ender 3

### Name abnormal settings

Tree supports. Slices fine with normal supports, enabling tree supports produces the error.

### Describe model location

Rotating it sometimes let it slice but mostly did not

### Describe your model

It is manifold. It slices fine with normal supports, just not tree supports.

### Add your .zip here ⬇️

[CE3E3V2_Corner Plug.zip](https://github.com/Ultimaker/Cura/files/13757005/CE3E3V2_Corner.Plug.zip)

5axes, 2023-12-23:
Interresting issue . The creation of supports is very uneven. slightly changing a parameter can give the right result, but the calculation is not very stable.

[image]

ThomasRahm, 2023-12-24:
Thank you for bringing these to my attention. I assumed the crashes were from me calling the infill class, but that is not the case.
Here we have a healthy mix of everything.

### These are actual bugs in the tree support (Will do a pull request soonish): 

https://github.com/Ultimaker/Cura/issues/17688 and https://github.com/Ultimaker/Cura/issues/17591 and https://github.com/Ultimaker/Cura/issues/17549 and https://github.com/Ultimaker/Cura/issues/17351
progress_inserts_check_interval variable should be larger than 0.

https://github.com/Ultimaker/Cura/issues/17670 and https://github.com/Ultimaker/Cura/issues/17529 and https://github.com/Ultimaker/Cura/issues/17479 and https://github.com/Ultimaker/Cura/issues/17331
There is more than a crash going on. Also definitely my bug but I will have to investigate this in more depth. 
Having an element on layer 0 that says it is resting on the buildplate, but not on a flat surface, which causes the code to try to drop it down (which crashes). Yeah it shouldn't crash, but something broke long before that.

### In Tree Support, but not caused by me:
https://github.com/Ultimaker/Cura/issues/17655  and https://github.com/Ultimaker/Cura/issues/17578
Only crashes when compiled as release with debug info. Does not crash if compiled in debug. Crash occurs in clipper. 

https://github.com/Ultimaker/Cura/issues/17597
Same crash as above but also crashes in debug mode. 

The whole vector is handed to the fillInfillParts function. I have the strong suspicion that it crashes if the layer above is modified while said function tries to access it. Can fix this.

###  Bugs, but not mine:
https://github.com/Ultimaker/Cura/issues/17412
If one changes to regular support, it also crashes.
Crashes in AreaSupport::detectOverhangPoints. Cant properly debug it as getting there when compiled in debug takes very long. Seems like part.outline is empty causing the part.outline.outerPolygon() Polygon to not have a path which causes part.outline.outerPolygon().area() to crash when dereferencing the path pointer. I suspect this is indicative of something else going on because if the crash here is prevented, it crashes later in the orderOptimizer.

###  Can not reproduce:
https://github.com/Ultimaker/Cura/issues/17582 and https://github.com/Ultimaker/Cura/issues/17575 and https://github.com/Ultimaker/Cura/issues/17521 and https://github.com/Ultimaker/Cura/issues/17520
Do not crash for me. I have the suspicion that these could be the same as https://github.com/Ultimaker/Cura/issues/17655 and https://github.com/Ultimaker/Cura/issues/17578 and https://github.com/Ultimaker/Cura/issues/17597 but do not run in the race condition on my device.

ThomasRahm, 2023-12-24:
Pull request: https://github.com/Ultimaker/CuraEngine/pull/2002

This should fix a log of the bugs linked above (Except https://github.com/Ultimaker/Cura/issues/17412 as it is not tree support related)

rburema, 2024-01-17:
Thomas' fix is merged now, will be in 5.7.x unless QA forces us to revert it (which I don't expect, quite the opposite).

( Internal: Also see jira ticket CURA-11519 )

MariMakes, 2024-01-19:
Even better news 😊, this fix is also included in the 5.7 Alpha, which has been released to analyze the incoming slicing crashes. 
If this crash is blocking you in any way you can download the nightly build here to continue printing (and to help us analyze the slicing crashes) https://github.com/Ultimaker/Cura/discussions/18080

Since we have a solution in place. I'll be closing this issue. 

Thanks for your report and Thomas' wonderful fix! Together we can make Cura better 😊

closed 2024-01-19 by MariMakes