Ultimaker/Cura #17744: Slicing failed with tree supports on 5_6

NEBERLIN, 2023-12-23:
### Cura Version

5.6

### Operating System

Windows 11

### Printer

Creality Ender 3 Pro Direct Drive Sprite Extruder

### Reproduction steps

added 2 copies of the model, added tree supports with default settings, hit slice, slicing failed,, ender 3 pro with sprite direct drive, using creality provided firmware from their website, normal supports work, tree supports fial

### Actual results

slicing failed

### Expected results

slicing completes

### Add your .zip and screenshots here ⬇️

[Cura5_6_bug.zip](https://github.com/Ultimaker/Cura/files/13757909/Cura5_6_bug.zip)

5axes, 2023-12-23:
We need an UltiMaker Cura project file for anyone to be able to help you on this issue.
guide: https://support.ultimaker.com/s/article/1667411030205

NEBERLIN, 2023-12-24:
Attached, will try to upload directly to github as well

NEBERLIN, 2023-12-24:
[CE3PRODD_06_Tool_Mount_(1)_GPLA_0.zip](https://github.com/Ultimaker/Cura/files/13760776/CE3PRODD_06_Tool_Mount_.1._GPLA_0.zip)

NEBERLIN, 2023-12-24:
added ultimatker project file