Ultimaker/Cura #17745: Multiple models printing One At A Time crashing print head after first model

flyingcheesehead, 2023-12-23:
### Cura Version

5.4.0

### Operating System

Mac OS Ventura 13.3

### Printer

Anycubic Kobra Plus, firmware 2.8.3

### Reproduction steps

Twice now I have put multiple models together and chosen to have them print One At A Time, and found the printer with the Z axis in a negative position, pushed so hard into the build plate that the plate is bent. One such file is attached, the other no longer exists.

### Actual results

This seems to happen after the first model. I was asleep the first time and my wife stopped the print after hearing terrible noises. This second time, I was only one floor away and started to hear unusual noises and ran down, but there was enough plastic around the hot end that it must have been extruding directly onto the bed and squishing out for a while:

[image]

Whatever is happening, the printer doesn't seem to realize that the head is jammed into the build plate, and tries to keep everything running as normal, which has caused (repairable) damage to the printer both times.

### Expected results

A successful print of all models in the file.

### Add your .zip and screenshots here ⬇️

Cura project file:
[Failed between model 1 and 2 of 5 AKMAX_Side Bolt x 2.3mf.zip](https://github.com/Ultimaker/Cura/files/13758370/Failed.between.model.1.and.2.of.5.AKMAX_Side.Bolt.x.2.3mf.zip)

GCode file that was generated and crashed the print head:
[AKMAX_Side Bolt x 2.gcode.zip](https://github.com/Ultimaker/Cura/files/13758374/AKMAX_Side.Bolt.x.2.gcode.zip)

Asterchades, 2023-12-24:
One at a Time mode requires you to have correctly defined the size of your tool head in the `Machine Settings`. This is the only way it knows where it's safe to position the various models to avoid any collisions - if that information is wrong, it's going to provide the wrong clearance when arranging the models.

The project you've given has the tool defined as being 30mm in X and 20mm in Y, offset 20mm to the left and 10mm behind the nozzle, with a 350mm high gantry. This has been inherited from the base Cura definition, rather than having been specifically defined by the Kobra Max definition, so is extremely unlikely to be incorrect. I don't know the exact dimensions of the Kobra Max tool head but it looks like it's going to be larger than that, evenly spaced left-to-right, and the gantry definitely isn't that far above the nozzle tip.

flyingcheesehead, 2023-12-25:
I just realized that Cura apparently thinks the build plate is bigger than it is. This is an Anycubic Kobra Plus (first generation), which is pretty much the same as the Vyper and the Kobra Max and is between those two in size, with a 300x300x350 build volume. However, looking at the size of the first part that printed in the picture above and comparing it with what is on the screen in Cura, it appears that Cura thinks my build volume is larger than it is, even though it is set correctly in the machine settings. This appears to be the root of the problem. Screen shot of machine settings below.

[image]

flyingcheesehead, 2023-12-25:
> One at a Time mode requires you to have correctly defined the size of your tool head in the `Machine Settings`. This is the only way it knows where it's safe to position the various models to avoid any collisions - if that information is wrong, it's going to provide the wrong clearance when arranging the models.
> 
> The project you've given has the tool defined as being 30mm in X and 20mm in Y, offset 20mm to the left and 10mm behind the nozzle, with a 350mm high gantry. This has been inherited from the base Cura definition, rather than having been specifically defined by the Kobra Max definition, so is extremely unlikely to be correct. I don't know the exact dimensions of the Kobra Max tool head but it looks like it's going to be larger than that, evenly spaced left-to-right, and the gantry definitely isn't that far above the nozzle tip.

Yes, I believe you are correct, the gantry is definitely not 350mm, that's the top of the build volume. Except I'm not sure that's what's causing the problem. The head isn't hitting another model, and like I just posted, looking at the placement of the objects in my picture it seems that Cura thinks the bed is bigger than it is defined in the settings.

Asterchades, 2023-12-25:
You're right that something appears to be amiss in that first image, in that it would seem to be that the first piece ("Door Frame Cap") is printing in the wrong place at the rear right instead of the front right. However it is not the result of Cura assuming your print volume is larger than specified - it's only using the 300 x 300, and the provided GCode confirms that as well. Per the provided Gcode, the skirt going around that piece would fit in a box from (145.239, 1.469) to (296.853, 190.845), which is indeed the front right of the bed.

While I haven't manually parsed the remainder of the file, loading it into two separate 3rd party GCode viewers also doesn't show anything being positioned beyond the limits of a 300 x 300 build plate. There also appears to be nothing that would inadvertently reset the Y coordinate, such as a G92. This tells me that the code generated is good, and as a result the issue lies outside of Cura.

Unfortunately this leaves me at a loss as to what caused your print to start so far to the rear. My first instinct would be to ensure the bed has smooth, uninterrupted travel across the full length - the Kobra Max uses sensorless homing, and will treat sufficient resistance as having reached a stop.

flyingcheesehead, 2023-12-26:
> You're right that something appears to be amiss in that first image, in that it would seem to be that the first piece ("Door Frame Cap") is printing in the wrong place at the rear right instead of [...]

That's good information, thank you! I think I solved the placement issue - One of the cables to the bed was in a spot where it could be pushed into the Y axis stop switch. If that happened when homing at the beginning of the print, the bed would have started in the wrong place.

It seems that the only remaining question is why the Z axis is going negative and trying to shove the nozzle through the bed. It clearly knows the correct spot at the beginning since the first model prints fine. Is there anything in the GCode that makes it try to go down more than it was up, or use the leveling sensor to rehome the Z axis or something? Thanks!

Asterchades, 2023-12-27:
The lowest Z value ever referenced in the provided GCode is 0.2, which is occurring the expected five (5) times. There's no attempts to re-set the 0 point beyond the initial G28 homing command that occurs before any printing is done. Once more the code appears to be good, which puts the issue outside of Cura's control.

I'm afraid I have much less of an idea as to what might cause this behaviour. Best guess in this case is either a loose Z coupler (gravity could help it drop further than it raises) or the ABL map trying to correct for a low spot which doesn't exist where the bed was actually positioned at the time.

GregValiant, 2025-01-15:
I'll go ahead and close this.  Let me know if that is premature.

closed 2025-01-15 by GregValiant