Ultimaker/Cura #17750: i get an unexpected error when trying to slice

toshimaro, 2023-12-23:
### Cura Version

5.6.0

### Operating System

windows11

### Printer

anycubic kobra neo

### Describe your model

no

### Add your .zip here ⬇️

[unexpextederrorproject.zip](https://github.com/Ultimaker/Cura/files/13759623/unexpextederrorproject.zip)

5axes, 2023-12-23:
An other "Print thin walls" issue : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452
[image]
or with tree support.
[image]

Duplicate : Slice Failed due to **Print Thin Walls**

https://github.com/Ultimaker/Cura/issues/17725
https://github.com/Ultimaker/Cura/issues/17679
https://github.com/Ultimaker/Cura/issues/17676
https://github.com/Ultimaker/Cura/issues/17671
https://github.com/Ultimaker/Cura/issues/17606
https://github.com/Ultimaker/Cura/issues/17560
https://github.com/Ultimaker/Cura/issues/17528
https://github.com/Ultimaker/Cura/issues/17503
https://github.com/Ultimaker/Cura/issues/17499
https://github.com/Ultimaker/Cura/issues/17470
https://github.com/Ultimaker/Cura/issues/17467
https://github.com/Ultimaker/Cura/issues/17457
https://github.com/Ultimaker/Cura/issues/17456
https://github.com/Ultimaker/Cura/issues/17496
https://github.com/Ultimaker/Cura/issues/17478
https://github.com/Ultimaker/Cura/issues/17439
https://github.com/Ultimaker/Cura/issues/17438
https://github.com/Ultimaker/Cura/issues/17410
https://github.com/Ultimaker/Cura/issues/17367
https://github.com/Ultimaker/Cura/issues/17344
https://github.com/Ultimaker/Cura/issues/17177
https://github.com/Ultimaker/Cura/issues/16688
https://github.com/Ultimaker/Cura/issues/16515
https://github.com/Ultimaker/Cura/issues/16417

toshimaro, 2023-12-23:
How can I slice without encountering an "unexpected error"?

5axes, 2023-12-23:
This error is due to the parameter "Print Thin Walls" . You just have to unset this parameter.

toshimaro, 2023-12-23:
[image]
I've unset "Print Thin Walls," but I'm still getting an "unexpected error" when slicing. Why is this happening?

5axes, 2023-12-23:
Could be an other issue as you are running Cura under Windows 11. Many user have also reprorted some issue on this OS.  You will have to wait for a deaper analyse from the developpent team.

toshimaro, 2023-12-24:
I am having trouble locating the 'Print Thin Walls' parameter in Cura. I have tried looking under the Custom settings but cannot find it. Can someone guide me to the exact location of this setting in the latest version of Cura? Any help would be greatly appreciated. Thank you!

toshimaro, 2023-12-24:
thx i found it!!

MariMakes, 2024-01-18:
Hi @toshimaro,

Quick update from our side 👋
We are introducing a new tool to analyze slicing crashes like the one you reported. 💪

Even with the changes we made in the latest versions of Cura we still see that your model crashes. 🫤 But we have added your project file to our analyzing tool and found there were at least 10 crashes that were similar to yours, which helps us to narrow down what the cause is of your crash so we can resolve this for you and others.
Hopefully, we can find a solution soon 🤞

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/17438

closed 2024-01-18 by MariMakes