Ultimaker/Cura #17751: Scaled up model with tree support in center of build plate does not slice

ChrisZehr, 2023-12-23:
### Cura Version

5.5.0

### Operating System

Windows 10, RTX 4080

### Printer

Anycubic Vyper

### Name abnormal settings

Scaled up 500%

### Describe model location

Moving the model to the front of the build plate allows it to slice without error

### Describe your model

M1A1 Abrams body

### Add your .zip here ⬇️

[AV_M1A1_Abrams_w Tree Support.zip](https://github.com/Ultimaker/Cura/files/13759629/AV_M1A1_Abrams_w.Tree.Support.zip)

5axes, 2023-12-23:
Don't know if we have to classify in Another "Part Position issue" or "Tree Support Issue". Anyway rotate the model solved the issue : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-6793499

[image]

ChrisZehr, 2023-12-24:
> (comment)

Did you try moving it around the build plate as well? It seemed to have an
issue in random locations when I tried to slice. It would either hang at
50% or give me the error message or work. I couldn't get it to be
consistent. Maybe it's fixed in 5.6.0?

closed 2024-04-03 by HellAholic