Ultimaker/Cura #17752: [5.6.0] Skirt has multiple layers!!

Sophist-UK, 2023-12-23:
### Cura Version

5.6,0

### Operating System

Windows 10

### Printer

Dagoma DiscoEasy 200

### Reproduction steps

Asked for a skirt -  line count 2, distance (from model) 2mm, min length 1mm.

### Actual results

Skirt is sliced, and has correct number of lines, but the innermost skirt line is printed on 3 layers not just the first layer.

### Expected results

Skirt is printed only for the first layer and not for subsequent layers.

### Add your .zip and screenshots here ⬇️

[image]

Asterchades, 2023-12-24:
Yes, yes it does.

With the release of Cura 5.4, a new feature was added to enable the Skirt to be multiple layers high, defaulting to three (3) layers. The intent is for it to be easier to remove. The actual setting is called, simply enough, `Skirt Height` and by default is not shown until Expert is selected or the setting is made visible manually; though you can always search for it. Setting this to one (1) will make it behave as it did prior to Cura 5.4.

Ref: https://github.com/Ultimaker/Cura/releases/tag/5.4.0
> - Introduced Skirt Height setting to make a skirt easier to remove

closed 2023-12-24 by Asterchades

Sophist-UK, 2023-12-24:
Ah - I will try it and see if it makes it easier to remove as promised.