Ultimaker/Cura #17766: Cura 5.6.0 Randomly Failing to Slice

Blarnix, 2023-12-24:
### Cura Version

5.6.0

### Operating System

Windows 11 22H2 Build 22621.2861

### Printer

Creality Ender 3

### Name abnormal settings

Acceleration/Jerk Control, Bridging settings, Infill travel optimizations, firmware compensation, arcwelder

### Describe model location

Center of buildplate. And No.

### Describe your model

Mesh fixes don't work. Happens with multiple models with no predictor.

### Add your .zip here ⬇️

[CuraErrorProjectFile.zip](https://github.com/Ultimaker/Cura/files/13762426/CuraErrorProjectFile.zip)

Blarnix, 2023-12-24:
It may be related to support structures, as toggling them _sometimes_ allows the model to be sliced. Can upload logs if need be.

papaof2, 2024-03-14:
Cura 5.6.0  Win 10 64 bit, quad core processor, 8GB RAM - everything else is working OK - browser, email.  Just installed Cura, took it's defaults for an Anet A8 printer and it's worked fine on several other files.  However Cura has been chewing on AA8_BeltPulleyInsert_v1_Anet_A8.stl for more than an hour.   The file is 852,655 bytes and I know some of the other files I tested with were much larger.  Task manager shows Cura is using about 13% of the CPU, 130 MB of RAM and is switching between High and Very High power usage.   

Bug in