Ultimaker/Cura #17769: Unable to import profiles from 5.3 into 5.6

nh-eng, 2023-12-24:
### Cura Version

5.6

### Operating System

Windows 11

### Printer

Elegoo Neptune 3 MAX

### Reproduction steps

Try to import all of my profiles from V5.3

### Actual results

The system attempts to load profiles from C:\Users\paul\Documents\Projects\3D_Printing_tools where there is a single file with a .curaprofile file extension. Searching my hard drive, I see that this is the only file with this extension, yet I have about a dozen profiles. It seems that 5.6 does not know where to look to import my profiles. I do see files in the Vx.x/quality_changes folders with .inst.cfg file extensions that are named the same as my profiles, but pointing Cura 5.6 does not display any supported files, so I am able to import them

### Expected results

Should load all of my previous profiles

### Add your .zip and screenshots here ⬇️

[cura.zip](https://github.com/Ultimaker/Cura/files/13762863/cura.zip)

blake-dutra, 2024-01-05:
I'm having the same issue, tried 5.5 as well and had the same problem. I guess I'm sticking with 5.3 for now.

fieldOfView, 2024-01-07:
> The system attempts to load profiles from C:\Users\paul\Documents\Projects\3D_Printing_tools

What makes you think that?

When a new version of Cura is started for the first time (or actually, when there is no configuration folder matching the version of Cura you are running inside `%APPDATA%\cura`), Cura looks inside `%APPDATA%\cura` for the latest version of a configuration folder, and make a copy of that with the current. The configuration folder does not use .curaprofile files, they are only make for end-user convenience, and it should not be necessary to use them to copy setting from an older version of Cura to the newer version.

nh-eng, 2024-01-07:
The reason I believe that is because there is a file there with a
.curaprofile extension.