Ultimaker/Cura #17770: Extruder retracts uncontrollably after stopping a print in the middle of the print job.

rbenites, 2023-12-24:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Creality Ender 3 S1 Pro with Sonic Pad

### Reproduction steps

1. Slice any print through the CURA slicer.
2. Send the print to the printer.
3. Start the print job.
4. In the middle of the print, STOP the print.
5. Clear the bed and start the print again.
6. Notice the extruder will start retracting wildly (at full speed for about 5 seconds) when performing the purge action.

### Actual results

After having previously stopping a print in the middle of the job, the extruder will retract at full speed for about 5 seconds until the extruder freezes when trying to print again.

### Expected results

The extruder should not retract wildly when trying to print.

### Add your .zip and screenshots here ⬇️

[CFFFP_Christmas_Stitch_Ornament.zip](https://github.com/Ultimaker/Cura/files/13762870/CFFFP_Christmas_Stitch_Ornament.zip)

Asterchades, 2023-12-25:
This sounds like your START_PRINT macro does not include anything to reset the E value while still trying to use absolute extrusion values. As a result it's maintaining the E as it was when you stopped the previous print, and when it tries to purge use a lower E value it will retract. Half-way through this file, for example, the E value reaches around 1,313, so if your first purge line tries to use E5 (for example) it will become a 1,308mm retraction instead of a 5mm extrusion.

You either need to incorporate a `G92 E0` into your START_PRINT macro, or make sure one occurs before it during your Start GCode.

Of course without knowing the actual contents of your START_PRINT macro I cannot say anything with 100% certainty. I'm fairly confidant that this is the issue, however.

Dirk-76, 2024-01-07:
I reported the same problem 2 weeks ago
[https://github.com/Ultimaker/Cura/issues/17764#issue-2055126245](url)
and it happen in the middle of the print.

Here my Start GCode
```
M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z20.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate
G92 E0 ; Reset Extruder
G28 ; Home all axes
G29 L0 ; Load mesh from slot 0 (or use any other previously saved slot)
G29 A ; Activate UBL
M420 S1 Z2; Use a mesh leveling up to 2mm (a valid mesh must be loaded into RAM);
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish

```
Looks like its somthing else.

Asterchades, 2024-01-07:
Doing the same thing doesn't mean it's the same issue - especially if it's occurring at a completely different place.

Without a _full copy_ of your resulting GCode, @Dirk-76, there's absolutely zero way to diagnose what's causing your problem. @rbenites' issue is very clearly the result of failing to reset the E location before attempting any print moves (which are configured in a macro that has not been provided), where I'd wager yours is the result of a post processing script being inserted that is using relative E values while the printer remains configured for absolute E values. That it works on older versions will be because the same script is simply not active.

Note: Please don't co-opt this post should you decide to provide the necessary information. Post it in your existing thread, then ping me (eg @Asterchades) and I'll take a look.