Ultimaker/Cura #17773: Vase mode end over-squeezing

Gornerrrrr, 2023-12-25:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

DIY FDM

### Reproduction steps

1. Fine quality
2. Height= 1.0 mm, Width= 2.0mm
3. Top layer= 0, Bot layer= 0, Build plate= None
4. Chosse "Spiralize Outer Contour"
5. Add a cube test model
6. Slice

### Actual results

At the end of the spiral pattern, there is a noticeable downward path, where the Z-axis coordinates not only fail to smoothly transition to the final layer but also experience a descent.

> ;TIME_ELAPSED:93.949133
;LAYER:18
;TYPE:WALL-OUTER
;MESH:Cube
G1 X41 Y41 Z19.25 E434.56359
G1 X59 Y41 Z19.5 E440.2073
G1 X59 Y59 Z19.75 E445.85101
G1 X41 Y59 Z20 E451.49472
;TIME_ELAPSED:98.952636
;LAYER:19
;TYPE:WALL-OUTER
;MESH:Cube
G1 F1028.6 X41 Y41 Z20.125 E457.13802
G1 X59 Y41 Z20.25 E462.78133
G1 X59 Y59 Z20.375 E468.42463
G1 X41 Y59 Z20.5 E474.06793
G1 X41 Y41 Z20 E478.30194 ;down 0.5mm over-squeezing
G1 X59 Y41 E481.12352
G1 X59 Y59 E482.53432
G1 X41 Y59
;TIME_ELAPSED:107.369619

### Expected results

In the old versions of Cura, there will no longer be a change in height when spiraling to the final height.

> ;TIME_ELAPSED:93.949133
;LAYER:18
;TYPE:WALL-OUTER
;MESH:Cube
G1 X41 Y41 Z19.25 E434.56359
G1 X59 Y41 Z19.5 E440.2073
G1 X59 Y59 Z19.75 E445.85101
G1 X41 Y59 Z20 E451.49472
;TIME_ELAPSED:98.952636
;LAYER:19
;TYPE:WALL-OUTER
;MESH:Cube
G1 F1028.6 X41 Y41 Z20.125 E457.13802
G1 X59 Y41 Z20.25 E462.78133
G1 X59 Y59 Z20.375 E468.42463
G1 X41 Y59 Z20.5 E474.06793
G1 X41 Y41 E478.30031
G1 X59 Y41 E481.12189
G1 X59 Y59 E482.53268
G1 X41 Y59
;TIME_ELAPSED:107.357350

Also, I would like to ask if Cura has plans to update the flow control at the beginning and end of the spiral mode? This would be useful for large nozzle [diameters.](https://github.com/Ultimaker/Cura/issues/13042)

### Add your .zip and screenshots here ⬇️

[cubeTest.zip](https://github.com/Ultimaker/Cura/files/13763568/cubeTest.zip)

Gornerrrrr, 2023-12-25:
During actual usage, I sliced and printed a vase using the "vase mode." However, there was a severe and noticeable over-extrusion at the very end, prompting me to manually stop the process during the final stage.

smartavionics, 2023-12-25:
Hi @Gornerrrrr , see https://github.com/smartavionics/Cura/releases for a Cura 4 variant that does what you're asking. It tapers the flow at the top and bottom of the spiral. Hope this helps.

Gornerrrrr, 2023-12-25:
@smartavionics Thanks reply! This version is useful for me. But the new version is not just a matter of flow control; there is also a situation of Z-axis down. I believe this is a bug.

smartavionics, 2023-12-25:
Ah, yes, I see that down movement at the end. You're right, it must be a bug.

HumbleDeer, 2023-12-27:
> @smartavionics Thanks reply! This version is useful for me. But the new version is not just a matter of flow control; there is also a situation of Z-axis down. I believe this is a bug.

This could be intentional behaviour. The flowrate tapers off at the end as well, which can be observed by calculating the increase in extruder position per upwards shift in the Z direction. The last couple commands at -0.5 Z relative to the highest point have a starkly lower flowrate for those given sections. Not going down could possibly cause the line to not stick where you'd want it to, as there's now no longer enough filament.

Gornerrrrr, 2024-01-04:
@HumbleDeer Thanks advice! I have noticed a reduction in flow in the final layer, but I believe the descent of the Z-axis is an unnecessary operation, especially during printing with a large nozzle diameter. 
I am currently using a 3.0mm nozzle, with layer height 2.0mm + line width 3.0mm for printing. At the end of the spiral printing, there will be a very noticeable Z-axis downward movement of 1.1mm (50% layer height?), causing damage to the top of the printed model.

> G1 X149.384 Y150.638 Z201.074 E54.55831
G1 X150.101 Y150.117 Z201.078 E54.66411
G1 X150.567 Y149.743 Z201.08 E54.73543
G1 X151.286 Y149.082 Z201.085 E54.85201
G1 X151.938 Y148.373 Z201.089 E54.96699
G1 X152.661 Y147.417 Z201.094 E55.11007
G1 X153.327 Y146.196 Z201.1 E55.27609
G1 X153.734 Y145.07 Z201.106 E55.419
G1 X153.954 Y144.184 Z201.11 E55.52798
G1 X154.119 Y143.152 Z201.114 E55.65273
G1 X154.208 Y142.866 Z200 E55.78354 ; Z down this
G1 X154.441 Y142.41 E55.84161
G1 X154.805 Y141.975 E55.90593
G1 X155.102 Y141.729 E55.94966
G1 X156.135 Y141.167 E56.08301
G1 X157.097 Y140.674 E56.20559
G1 X157.865 Y140.302 E56.30236
G1 X158.351 Y140.168 E56.35953
G1 X158.724 Y140.133 E56.40201
G1 X159.125 Y140.164 E56.44762
G1 X160.328 Y140.445 E56.58771
G1 X161.695 Y140.726 E56.74597

[image]
[image]