Ultimaker/Cura #17786: Cura 5.6.0 has become unusable suddenly

Supatrent, 2023-12-26:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Creality Ender 3 Max

### Reproduction steps

I'm trying to slice the same files that I have done before so I know they should work. When I load the files onto the build-plate in "prepare" I get shadows under them that I can not remove. when I merge the files into one piece the shadows merge and make looking at the part from underneath impossible. 

I have merged parts and click on the "select face to align to build-plate it no-longer works and just unclicks the item. I have tried this on other files and it does it there also. I have restarted the program, restarted the computer, and re-installed Cura but it keeps happening.

### Actual results

Models have an obscuring shadow under them that I cannot remove or see the piece through.  I can not click select face to align to buildplate on any piece that has been merged.

### Expected results

I would like to be able to see the pieces and align faces to the buildplate 

### Add your .zip and screenshots here ⬇️

[Project file 1.zip](https://github.com/Ultimaker/Cura/files/13773359/Project.file.1.zip)
[image]
[image]

HumbleDeer, 2023-12-27:
Check whether disabling any kind of bed adhesion print helps. This looks like a preview of the skirt or brim being stuck in your model viewer.

Supatrent, 2023-12-27:
> Check whether disabling any kind of bed adhesion print helps. This looks like a preview of the skirt or brim being stuck in your model viewer.

I set the build plate adhesion to none, that made the shadow disappear under the buildplate.

Supatrent, 2023-12-28:
Additional Information:

I downloaded an older version (5.5.0) and sliced fine for a day, then just a few minutes ago the same two above issues happened.