Ultimaker/Cura #17787: Slicing Failed on Goku Print

CreativeSpaceManufacturing, 2023-12-26:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Custom - Voxelab Aquilla X2

### Name abnormal settings

Yes it does

### Describe your model

Slices fine with preset setting. Doesn't like my print profile that has worked on other models.

### Add your .zip here ⬇️

[Goku 200mm Tall.zip](https://github.com/Ultimaker/Cura/files/13773732/Goku.200mm.Tall.zip)

5axes, 2023-12-28:
An other Windows 11 issue : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

[image]

Duplicate : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well?
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic

CreativeSpaceManufacturing, 2024-04-03:
Hello,

I was able to print it but I had to turn off the ironing feature. That
change allowed it to slice.
But I will also see if I am able to do it with the newest version of Cura.