Ultimaker/Cura #17789: Sovol SV04 Single Mode 02

RRoyce23, 2023-12-26:
### Cura Version

5.6

### Operating System

Window 10

### Printer

Sovol SV04

### Reproduction steps

Tried to slice and print only using right extruder (#2). The software does not seem to recognize the #2 extruder in this mode. Further the filament diameter is set to 2.85 vs the 1.75 that it should be. When attempting to print, the printer uses #1 extruder. The Sovol Cura software accurately recognizes the #2 extruder.

### Actual results

Uses wrong extruder

### Expected results

There is no option to select the right (#2) extruder ... it simply gives an option to select an extruder and keeps the default #1 extruder.

### Add your .zip and screenshots here ⬇️

[SSSM02_Handles.zip](https://github.com/Ultimaker/Cura/files/13773793/SSSM02_Handles.zip)

DrewCPU, 2024-02-04:
I think I resolved this by adding `T1` to the beginning of my start GCODE. I don't believe there was anything to tell it to use any extruder than the one it was last set to.

GregValiant, 2024-11-20:
That's the basic fix for the problem.  You might want to include "M605 S0" right before the T1 in order to set the "mode" in the printer.
I'll go ahead and close this as it is a printer definition problem (Sovol problem) rather than a Cura bug.

closed 2024-11-20 by GregValiant