Ultimaker/Cura #17792: Wipe move goes in wrong direction in Cura 5.4 and newer

DerKob, 2023-12-26:
### Cura Version

5.4.0 & 5.6.0

### Operating System

Windows 10

### Printer

Artillery Sidewinder X1

### Reproduction steps

I slice a cylindrical object and look at the preview. You can see the movement of the nozzle at the Z-seam. If you set the wiping distance to 1 mm, you can see the bug more clearly. 

### Actual results

The wiping movement goes in the wrong direction. Instead of wiping over the Z-seam, the movement goes to the Z-seam and then back. This results in a poorer Z-seam instead of a better Z-seam.
[image]
[image]

The bug only occurs with some objects. I have also cut objects with CURA 5.6. where the wipe goes in the right direction. 

### Expected results

In Cura 4.13.1, wiping is always correct. It goes over the Z-seam and makes a better quality out of it. 
[image]
[image]

### Add your .zip and screenshots here ⬇️

I can not give you the file because it is from my company.

Khronos666, 2024-02-15:
the topic is still relevant, I have long pointed out this error... moreover, it turned out to be very critical...
I would like to know if it will be fixed at all?

https://github.com/Ultimaker/Cura/issues/15820

Khronos666, 2024-02-20:
> тема по-прежнему актуальна, я давно указывал на эту ошибку... более того, она оказалась очень критичной... Хотелось бы знать, будет ли она вообще исправлена?
> 
> #15820
I found an alternative ... everything works there... OrcaSlicer