Ultimaker/Cura #17801: Slicing Failure high rate Ender 3 V3 SE, only non-standard setting is Tree Supports being turned on

bbqxx, 2023-12-27:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Ender 3 V3 SE

### Name abnormal settings

Tree Supports is enabled, that's about it

### Describe model location

90 degrees, 2 of them, tried to center them close

### Describe your model

Sliced before without issue for Prusa, Ender 3, Ender 3 S1, and other printers without issue

### Add your .zip here ⬇️

[Bug Issues Cura.zip](https://github.com/Ultimaker/Cura/files/13780747/Bug.Issues.Cura.zip)

I have experience slicing models on Cura for a few years on many other printers, however I seem to be having issues slicing about 60% of the time, regardless of orientation, when slicing specifically on 5.6.0 for the Ender 3 V3 SE printer settings. I tried keeping them stock settings, only allowing for Tree supports but regardless of what I try to do, it keeps generating the same error "Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker."

At first I thought it was just a 1-off issue, but after encountering the issue on 21/32 models, multiple times, some of which regardless of what I attempt, I keep encountering the same issue. I have tried re-downloading Cura in case it was an issue due to a bad download, but nothing has changed, so I'm reporting this bug (I know it's not a lot of information to go off of) in hopes it can be rectified.

5axes, 2024-01-04:
No issue for me on Windows 10 : 
[image]

An other Windows 11 issue : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

Duplicate Slice Failed on **Windows 11**

https://github.com/Ultimaker/Cura/issues/17895
https://github.com/Ultimaker/Cura/issues/17818
https://github.com/Ultimaker/Cura/issues/17801
https://github.com/Ultimaker/Cura/issues/17798
https://github.com/Ultimaker/Cura/issues/17790
https://github.com/Ultimaker/Cura/issues/17787
https://github.com/Ultimaker/Cura/issues/17775
https://github.com/Ultimaker/Cura/issues/17675
https://github.com/Ultimaker/Cura/issues/17659
https://github.com/Ultimaker/Cura/issues/17645
https://github.com/Ultimaker/Cura/issues/17636
https://github.com/Ultimaker/Cura/issues/17635
https://github.com/Ultimaker/Cura/issues/17617
https://github.com/Ultimaker/Cura/issues/17613
https://github.com/Ultimaker/Cura/issues/17612
https://github.com/Ultimaker/Cura/issues/17595
https://github.com/Ultimaker/Cura/issues/17571
https://github.com/Ultimaker/Cura/issues/17567
https://github.com/Ultimaker/Cura/issues/17566
https://github.com/Ultimaker/Cura/issues/17552
https://github.com/Ultimaker/Cura/issues/17546
https://github.com/Ultimaker/Cura/issues/17538
https://github.com/Ultimaker/Cura/issues/17530
https://github.com/Ultimaker/Cura/issues/17526
https://github.com/Ultimaker/Cura/issues/17523
https://github.com/Ultimaker/Cura/issues/17509
https://github.com/Ultimaker/Cura/issues/17491
https://github.com/Ultimaker/Cura/issues/17490
https://github.com/Ultimaker/Cura/issues/17461  ??
https://github.com/Ultimaker/Cura/issues/17380
https://github.com/Ultimaker/Cura/issues/17343
https://github.com/Ultimaker/Cura/issues/17335
https://github.com/Ultimaker/Cura/issues/17253

MariMakes, 2024-01-08:
Hey @bbqxx,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Thanks for the Project File 👍 I'm on Windows 11 and I took a look at it in Cura 5.6 and I've been able to slice it, so I don't believe that it's a windows 11 issue but something else related to your set-up 🤔
[image]

I'll bring it up to the team to see what they can do to improve it. 
Fingers crossed 🤞 

Thanks for your report 👍

closed 2024-04-03 by HellAholic