Ultimaker/Cura #17806: Align face to the build plate does not work

AlbertoGirardi, 2023-12-27:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Flying Bear Ghost 5

### Reproduction steps

1. Loaded an stl model
2. In the "rotate" section click "select face to align to the build plate"
3. click a flat place 

### Actual results

The model is rotated in a random orientation

https://github.com/Ultimaker/Cura/assets/122226906/ebfd6f9a-35b7-4096-838e-b7bd15b8a262

### Expected results

The model should be rotated with that face touching the build plate

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/Ultimaker/Cura/files/13782038/cura.log)

This could be a duplicate of #16035 but that issue was closed so I am creating this one

fieldOfView, 2023-12-28:
From the issue you linked to:

We'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting.
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the view it on GitHub link at the bottom of your email to open GitHub in the browser.

AlbertoGirardi, 2023-12-28:
[cura_bug (2).zip](https://github.com/Ultimaker/Cura/files/13789419/cura_bug.2.zip)

While trying to reproduce the bug today the issue wasn't present at first. I then closed cura and restarted it, and the issue returned.