Ultimaker/Cura #17811: Slicing with tree support fails

Hari-B6, 2023-12-28:
### Cura Version

5.6.0

### Operating System

Windows 11 23H2 Build: 22635.2915 (Windows Insider)

### Printer

Ultimaker Original (without heatplate upgrade)

### Name abnormal settings

support is set to tree

### Describe model location

moving or rotating the model doesn't fix the issue

### Describe your model

yes, with normal support structure

### Add your .zip here ⬇️

[Archive with the .stl and .mf project files](https://github.com/Ultimaker/Cura/files/13788544/Project.zip)

5axes, 2024-01-04:
An other duplicate Tree Support issue : https://github.com/Ultimaker/Cura/issues/17741#issuecomment-1868413301
Actual list : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452
[image]

Vandresc, 2024-01-19:
👋 Quick update on our side.
Your issue seems to have been resolved with this fix from our wonderful community member ThomasRahm!
(https://github.com/Ultimaker/CuraEngine/pull/2002)
 

I'll close this issue since it's resolved, a fix should be available with the upcoming release. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2024-01-19 by Vandresc

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well?
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

Hari-B6, 2024-05-03:
I managed to print that model, and I haven't encountered any slicing problems since.