Ultimaker/Cura #17816: Slicing fails depending on tree positioning settings

Slashee-the-Cow, 2023-12-29:
### Cura Version

5.6

### Operating System

Windows 10 x64 22H2, GF1080Ti (Nvidia drivers 546.17)

### Printer

Ender-3 V3 SE, slightly customised profile

### Name abnormal settings

Changing tree support - "On Model If Required" works, "On Buildplate When Possible" fails

### Describe model location

Slices fine if I move it to the side (in this case X axis) a bit

### Describe your model

No errors when loading model. Exported from a FreeCAD file. Is a single watertight body (the Part Design workbench won't shut up if either of those are false)

### Add your .zip here ⬇️

[E3V3SES_PETG_filament_tube_holder_chasm.zip](https://github.com/Ultimaker/Cura/files/13792714/E3V3SES_PETG_filament_tube_holder_chasm.zip)
[filament tube holder chasm freecad.zip](https://github.com/Ultimaker/Cura/files/13792728/filament.tube.holder.chasm.freecad.zip)
Slicing failed again when changing print speed, but moving slightly again fixed that.

5axes, 2024-01-04:
Cannot really test as the machine "creality_ender3v3se_slashee" is not a standard machine. But looks like one of the Duplicate Tree Support issue : https://github.com/Ultimaker/Cura/issues/17741#issuecomment-1868413301
Actual list : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

Slashee-the-Cow, 2024-01-04:
> (comment)
> Actual list : #16440 (comment)

It's just a modification of the standard E3V3SE definition with slightly
different startup/end gcode and the correct size for print head and height
for the gantry (which aren't in the stock profile) to stop it knocking over
stuff in one at a time prints.

closed 2024-01-19 by Vandresc

MariMakes, 2024-01-19:
Eyyy @Slashee-the-Cow ❤️ 👋 

Even better news 😊, this fix is also included in the 5.7 Alpha, which has been released to analyze the incoming slicing crashes.
If this crash is blocking you in any way you can download the nightly build to continue printing (and to help us analyze the slicing crashes). It also includes faster slicing times for models with a lot of holes like an Eiffel Tower model

You can find the downloads here https://github.com/Ultimaker/Cura/discussions/18080
Will you let us know what you think/find and forward people you encounter who might benefit from it? 
Disclaimer: It's still an internal build so it hasn't passed our usual quality checks, but we are happy to hear what you think.