Ultimaker/Cura #17818: Slicing Failure - BB_Eyes

Don-Kent, 2023-12-29:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Creality Ender-5 S1

### Name abnormal settings

None, No

### Describe model location

Slices when Reloaded and with different orientation

### Describe your model

Sliced multiple times but it won't slice to requested orientation

### Add your .zip here ⬇️

[CE5S1_Eyes.zip](https://github.com/Ultimaker/Cura/files/13795592/CE5S1_Eyes.zip)

5axes, 2024-01-03:
An other Windows 11 issue  :  https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

[image]

Duplicate : Slice Failed on **Windows 11**

https://github.com/Ultimaker/Cura/issues/17798
https://github.com/Ultimaker/Cura/issues/17790
https://github.com/Ultimaker/Cura/issues/17787
https://github.com/Ultimaker/Cura/issues/17775
https://github.com/Ultimaker/Cura/issues/17675
https://github.com/Ultimaker/Cura/issues/17659
https://github.com/Ultimaker/Cura/issues/17645
https://github.com/Ultimaker/Cura/issues/17636
https://github.com/Ultimaker/Cura/issues/17635
https://github.com/Ultimaker/Cura/issues/17617
https://github.com/Ultimaker/Cura/issues/17613
https://github.com/Ultimaker/Cura/issues/17612
https://github.com/Ultimaker/Cura/issues/17595
https://github.com/Ultimaker/Cura/issues/17571
https://github.com/Ultimaker/Cura/issues/17567
https://github.com/Ultimaker/Cura/issues/17566
https://github.com/Ultimaker/Cura/issues/17552
https://github.com/Ultimaker/Cura/issues/17546
https://github.com/Ultimaker/Cura/issues/17538
https://github.com/Ultimaker/Cura/issues/17530
https://github.com/Ultimaker/Cura/issues/17526
https://github.com/Ultimaker/Cura/issues/17523
https://github.com/Ultimaker/Cura/issues/17509
https://github.com/Ultimaker/Cura/issues/17491
https://github.com/Ultimaker/Cura/issues/17490
https://github.com/Ultimaker/Cura/issues/17461  ??
https://github.com/Ultimaker/Cura/issues/17380
https://github.com/Ultimaker/Cura/issues/17343
https://github.com/Ultimaker/Cura/issues/17335
https://github.com/Ultimaker/Cura/issues/17253

Vandresc, 2024-03-01:
Hey,
We've implemented several fixes in hopes to eliminate most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue still persists.
Cheers!

closed 2024-03-01 by Vandresc