Ultimaker/Cura #17821: Slicing Failed Error

Jswens24, 2023-12-30:
### Cura Version

5.6.0

### Operating System

MacOS Sonoma 4.2.1 (23C71)

### Printer

Anycubic Kobra Max

### Describe your model

This is the first time i have tried to slice this model

### Add your .zip here ⬇️

[cover_comfortable.STL.zip](https://github.com/Ultimaker/Cura/files/13797081/cover_comfortable.STL.zip)

5axes, 2024-01-04:
Could be an other "Print Thin Walls" issue  

[image]

Vandresc, 2024-03-01:
Hey,
We've implemented several fixes in hopes of eliminating most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue persists.
Cheers!

closed 2024-03-01 by Vandresc