Ultimaker/Cura #17822: Small flat item slicing fail

LDDSurfer, 2023-12-30:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

ELEGOO Neptune 3 Pro

### Name abnormal settings

Tree supports

### Describe model location

rotated 90*

### Describe your model

No to all

### Add your .zip here ⬇️

[Ebony Dagger (Skyrim) - 3353876.zip](https://github.com/Ultimaker/Cura/files/13797499/Ebony.Dagger.Skyrim.-.3353876.zip)

LDDSurfer, 2023-12-30:
Only works with Fine settings for some reason

Vandresc, 2024-03-01:
Hey,
We've implemented several fixes in hopes of eliminating most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue persists.
Cheers!

closed 2024-03-01 by Vandresc