Ultimaker/Cura #17824: Ouverture fichier STL

dgburbjdhfb, 2023-12-30:
### Cura Version

5.6.0

### Operating System

Windows11

### Printer

Anycubic Kobra

### Reproduction steps

J'ai converti plusieurs fichiers SLDPRT en fichier STL. Puis j'ai essayé de les ouvrir l'un après l'autre sur Cura.

### Actual results

Lorsque j'essaye de les ouvrir, soit ils n'arrivent pas à s'ouvrir alors que le fichier n'est pas corrompu et qu'il contient bien un modèle, ou soit il s'ouvre et les dimensions sont automatiquement mises à 10000% et lorsque que je les remet à 100% il ne s'affiche pas.

### Expected results

J'aimerais pouvoir ouvrir mes fichiers normalement pour pouvoir les imprimer. 

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]
[entretoise 2.zip](https://github.com/Ultimaker/Cura/files/13797726/entretoise.2.zip)

5axes, 2024-01-03:
Le problème n'est pas avec Cura. Ceci n'est pas un fichier STL. le problème est au niveau de la conversion depuis SolidWorks.

dgburbjdhfb, 2024-01-05:
Oui, c'était bien sa. Merci

dgburbjdhfb, 2024-01-05:
Erreur au niveau de l'enregistrement de mes fichiers.

closed 2024-01-05 by dgburbjdhfb