Ultimaker/Cura #17828: Slicing Error

bwheeler3260, 2023-12-30:
### Cura Version

5.6.0

### Operating System

Window 10

### Printer

Anycubic Kobra Max

### Name abnormal settings

No abnormal settings.  Doesn't work under any profile I choose

### Describe model location

Moving it on the build plate or rotating it doesn't fix the problem

### Describe your model

Yes.  I have sliced it before in previous versions of Cura.  I cannot recall which one though.

### Add your .zip here ⬇️

[AKMAX_pred_battle_full_mask.zip](https://github.com/Ultimaker/Cura/files/13798601/AKMAX_pred_battle_full_mask.zip)

5axes, 2024-01-04:
An other *Print Thin Walls* issue : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452
[image]

MariMakes, 2024-01-18:
Hi @bwheeler3260,

Quick update from our side 👋
We are introducing a new tool to analyze slicing crashes like the one you reported. 💪

Even with the changes we made in the latest versions of Cura we still see that your model crashes. 🫤 But we have added your project file to our analyzing tool and found there were at least 8 crashes that were similar to yours, which helps us to narrow down what the cause is of your crash so we can resolve this for you and others.
Hopefully we can find a solution soon 🤞

I'll be closing this issue as a duplicate but you can follow the progress here https://github.com/Ultimaker/Cura/issues/16417

closed 2024-01-18 by MariMakes

HellAholic, 2024-02-16:
Small update,
Issue is resolved in CURA-11570 and will be part of the next Cura release.