Ultimaker/Cura #17829: Slice failed

omgwtf2, 2023-12-30:
### Cura Version

5.50

### Operating System

Windows 11

### Printer

Ender 5 plus

### Name abnormal settings

no

### Describe model location

no

### Describe your model

yes

### Add your .zip here ⬇️

Worked fine till now.
[ASX1_Thumb_-_1_x_2.docx](https://github.com/Ultimaker/Cura/files/13798638/ASX1_Thumb_-_1_x_2.docx)

5axes, 2024-01-04:
Not a word document but a 3MF project.
[image]

And an other Tree Support Issue : https://github.com/Ultimaker/Cura/issues/17741#issuecomment-1868413301
List : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

MariMakes, 2024-01-19:
Even better news 😊, this fix is also included in the 5.7 Alpha, which has been released to analyze the incoming slicing crashes.
If this crash is blocking you in any way you can download the nightly build to continue printing (and to help us analyze the slicing crashes) 

You can find the downloads here https://github.com/Ultimaker/Cura/discussions/18080