Ultimaker/Cura #17839: Custom settings for printer dimensions not going into effect

melsbacksfriend, 2023-12-31:
### Cura Version

5.6.0

### Operating System

Kubuntu 23.10 (NVidia 4070)

### Printer

Creality ender-3 v2 with crtouch, ender extender and mriscoc firmware

### Reproduction steps

1. select creality ender-3 v2 in the setup
2. change dimensions to match the ender extender 400xl

### Actual results

It still would only let me slice up to 235mm x 235mm x 250mm

### Expected results

Being able to slice larger models

### Add your .zip and screenshots here ⬇️

[Screenshots.zip](https://github.com/Ultimaker/Cura/files/13800652/Screenshots.zip)

Asterchades, 2024-01-01:
There are excluded areas in the default Ender-3 definition which are intended to stop the nozzle from colliding with the bed clips. You can see them as the darker grey areas in your screenshot. When you increase the size of the bed those zones still exist in the same location relative to the centre, causing the restriction you're seeing.

You'll need to switch to a different base definition. There's no appreciable difference between the Ender-3 and the Ender-3 Pro, so you can safely switch to the latter - which does not include the restricted areas as the original hardware never had anything to avoid.

Do be aware that increasing the print volume dimensions won't adjust the visible printer model in the UI. So the modelled build plate will still only occupy an area in the middle rather than the entire usable area. The result is simply that the overhanging area will be a slightly different shade of grey and it will have zero impact on the usable print volume.

melsbacksfriend, 2024-01-02:
That worked but now I'm encountering a different issue that others seem to be having and that is that tree supports aren't working.

Asterchades, 2024-01-02:
I would encourage you to start a new ticket for that, with new details. That way the team isn't having to dig through potentially unrelated threads. Do be aware that slicing issues like that are also best accompanied by a project file, otherwise there isn't anything to use for diagnostics.

Sometimes it can be that making small adjustments is enough to coax the system into working properly. Rotating the model, moving it (even a fraction of a millimetre), or adjusting the layer height - including the initial layer height - have all been known as work-arounds until such time that the main issue itself can be resolved.

closed 2024-01-02 by Asterchades