Ultimaker/Cura #17843: tree support fail

MBogballe, 2023-12-31:
### Cura Version

5.6.0

### Operating System

windows 11

### Printer

ender 5 plus / anet a8 plus / DIY

### Reproduction steps

can't make tree support on a simple test file works fine in 5.5.0
I have tried with a standard settings

### Actual results

can't make tree support on a simple test file works fine in 5.5.0
I have tried with a standard settings

### Expected results

can't make tree support on a simple test file works fine in 5.5.0
I have tried with a standard settings

### Add your .zip and screenshots here ⬇️

[Support Test Piece - 2823038.zip](https://github.com/Ultimaker/Cura/files/13801337/Support.Test.Piece.-.2823038.zip)

5axes, 2024-01-03:
We need an UltiMaker Cura project file for anyone to be able to help you with this.
guide: https://support.ultimaker.com/s/article/1667411030205

But could be an other Windows 11 issue ?
[image]

MBogballe, 2024-01-03:
Hi therehave created a file with the profile and subject.sometimes it wants to slice it, but if I then move it, it won't.other times it won't slice it to begin with, but when I move it and try several times, it will go in between. it is a bit frustrating that there are so many errors in cura.have been struggling with errors since 5.2.0 and they are errors that I cannot solve by changing anything or using other settings.  MVHMikkel BøgballeBottrupvej 498723 Løsning+45 22414666 Fra: 5axesSendt: 3. januar 2024 18:33Til: Ultimaker/CuraCc: MBogballe; AuthorEmne: Re: [Ultimaker/Cura] tree support fail (Issue #17843) We need an UltiMaker Cura project file for anyone to be able to help you with this.guide: https://support.ultimaker.com/s/article/1667411030205But could be an other Windows 11 issue ?image.png (view on web)