Ultimaker/Cura #17847: Cannot slice past v.5.4.0; Windows 11, multiple PCs

musicartist525, 2023-12-31:
### Cura Version

5.6.0

### Operating System

Windows 11 Home 10.0.22621 Build 22621

### Printer

FLsun V400

### Name abnormal settings

No.  It generates slicing error on each profile attempted.

### Describe model location

No.  Arrangement or positioning does not affect it.

### Describe your model

Model slices normally in v.5.4.0.  It is watertight.

### Add your .zip here ⬇️

[FV_seedling_5x5 no slicing.zip](https://github.com/Ultimaker/Cura/files/13801945/FV_seedling_5x5.no.slicing.zip)

I only use v.5.4.0, and print daily with numerous models.  All slice normally.  v.5.5.0 and 5.6.0 (and intermediate beta versions) all generate slicing errors or just stop slicing with no message.

Justmichaud, 2024-01-02:
I been having the same issues. Still waiting for Cura to respond.

GregValiant, 2024-01-02:
There is an ongoing problem between Windows 11 and Cura.  The Cura team has been collecting data but I don't know if any progress has been made on solving the problem.  I haven't had any problems with "won't slice" on my Windows 10 installation.  (Slicing can be pretty slow though.)