Ultimaker/Cura #17850: Creality Cloud Integration Plugin Failing to Install

Clobbyy, 2024-01-01:
### Cura Version

5.4.0

### Operating System

Mac OS 14.1.2 (23B92)

### Printer

Creality Ender-3 V3 SE

### Reproduction steps

Creality Cloud Integration doesn't show up in marketplace and when the file is dragged in from the Creality Website it fails to install after restart.

### Actual results

fails to Install

### Expected results

Plugin should install and appear in Extensions

### Add your .zip and screenshots here ⬇️

N/A

GregValiant, 2024-01-02:
Ultimaker didn't design or write that plugin.  It isn't in the Marketplace so has not been approved by Ultimaker.
Why do you consider this a UM Cura problem and not a Creality problem?

GregValiant, 2026-09-10:
I'll close this.

closed 2026-09-10 by GregValiant