Ultimaker/Cura #17853: Failed to slice dragon statue

DaWhiteXD, 2024-01-01:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Ultimaker S3

### Describe model location

Center, a bit pulled below the buildplate

### Describe your model

The model is a dragon statue, and it failed to slice

### Add your .zip here ⬇️

[UMS3_ChainiceDragon2.zip](https://github.com/Ultimaker/Cura/files/13804240/UMS3_ChainiceDragon2.zip)

5axes, 2024-01-04:
An other "Part position dependant" issue : https://github.com/Ultimaker/Cura/discussions/16440#discussioncomment-7790452

[image]

Rotate the model of 90° seems so solved this issue. Duplicate Slicing success can be Part position dependant 

https://github.com/Ultimaker/Cura/issues/17697
https://github.com/Ultimaker/Cura/issues/17682
https://github.com/Ultimaker/Cura/issues/17670
https://github.com/Ultimaker/Cura/issues/17610
https://github.com/Ultimaker/Cura/issues/17598
https://github.com/Ultimaker/Cura/issues/17494
https://github.com/Ultimaker/Cura/issues/17473
https://github.com/Ultimaker/Cura/issues/17466

DaWhiteXD, 2024-01-13:
Problem solved, thx @5axes

MariMakes, 2024-01-18:
Hi @DaWhiteXD 

Quick update from our side 👋
We are introducing a new tool to analyze slicing crashes like the one you reported. 💪
I'm glad to hear that your issue has been resolved. Thanks for the help with that @5axes 😍 

Even with the changes we made in the latest versions of Cura we still see that your model crashes with the original settings. 🫤 But we have added your project file to our analyzing tool and found there were at least 5 crashes that were similar to yours, which helps us to narrow down what the cause is of your crash so we can resolve this for you and others.

Hopefully, we can find a solution soon 🤞
You can follow the progress here: https://github.com/Ultimaker/Cura/issues/17679