Ultimaker/Cura #17854: New problem. Bed not heating, Nozzle not heating and Printing 8cm up

stevebright64, 2024-01-01:
### Cura Version

5.6

### Operating System

Mac Os

### Printer

Crealility 3 V2 2 Neo

### Reproduction steps

I'd welcome advice on something strange. I use Ulitmaker Cura, Its worked well. Printed the Gcode nicely on my Creatlity 3 vs 2 Neo.   3 Weeks ago added G29 to Start code to get it to auto level before each Print. It worked well, No problems, Starting yesterday, I converted an STL to GCode , and the printer started to '? print' straight away, without heating bed or nozzle and was moving around about 8cm above the bed.  When I used a gcode from weeks ago, that printed nicely. Incase it was the STL, I sliced it in the creality slicer , that GCODE printed ok.  I've then loaded the latest version of Ultimake Cura 5.6.0 and same problem? Any ideas?

### Actual results

Bed no Heating, Nozzle Not heating, Trying to pront straight away, but 8cm above bed. 

### Expected results

Bed to Heat, Nozzle to heat, To print a line at side then th egcode on the bed.

### Add your .zip and screenshots here ⬇️

No zip

GregValiant, 2024-01-01:
If you are printing from the SD card this sounds like a classic case of bad memory sectors on the card.  I have had good luck re-formatting memory cards or you can just try a different one.  If you buy a new card I would suggest keeping the size at 32mb or below.  Some Creality machines don't deal well with higher capacity cards.

apttie, 2024-01-02:
Sounds like it might be worth inspecting the 'bad' gcode.  Can you confirm that your [Startup G-Code from config] is unmodified in settings and made it into the 'bad' gcode correctly?  If you inspect the g code and find your start up code intact then maybe post the bad gcode (up until the beginning of the print)?

christi992, 2024-03-18:
I have also had this same issue happen to me lately. Never had a problem for over a year and now its not putting the startup gcode even though it has it on the machine settigs. The only thing different ive done is add the measure extension.

closed 2024-04-01 by github-actions[bot]