Ultimaker/Cura #17857: Material Manager causes immediate crash

LilBub, 2024-01-01:
### Cura Version

5.6

### Operating System

Windows 11

### Printer

Ultimnaker S5

### Reproduction steps

1 Open Cura
2 Hit Control K 
3 Wait 4 seconds
4 Scary error message
5 Click Send Report

### Actual results

[image]

### Expected results

I wanted to Duplicate the Generic PLA to add a new material

### Add your .zip and screenshots here ⬇️

[image]

fieldOfView, 2024-01-03:
Please post your cura.log file, which contains much more information than the screenshot you included.

LilBub, 2024-01-04:
I'm trying to attach the log file but I get it is telling me it does not support that type of file (txt)

fieldOfView, 2024-01-04:
From the logfile attached [here](https://community.ultimaker.com/topic/44784-56-blows-up-when-entering-material-manager/?do=findComment&comment=329189), and the subsequent testing by another user, it seems that this issue is related to the Material Settings plugin. Can you test if disabling that plugin fixes the issue? If it does, please post an issue here:
https://github.com/fieldOfView/Cura-MaterialSettingsPlugin/issues

fieldOfView, 2024-01-07:
From the discussion on the community forum (https://community.ultimaker.com/topic/44784-56-blows-up-when-entering-material-manager/) it follows that this issue is caused by the Material Settings plugin. Therefore I am closing this issue on the main Cura repository, since this is something that needs to be fixed in the Material Settings plugin instead of in Cura.

closed 2024-01-07 by fieldOfView