Ultimaker/Cura #17862: Slicing fails with Sketchup .STL files.

djardi, 2024-01-02:
### Cura Version

Cura 5.6.0

### Operating System

Windows 11

### Printer

Ultimaker Cura 5.6.0

### Name abnormal settings

No changes

### Describe model location

No

### Describe your model

It slices without problem in Cura 5.4.0. Sketchup diagnostic shows no problems.

### Add your .zip here ⬇️

[CCR10S_FltLte_1.zip](https://github.com/Ultimaker/Cura/files/13808145/CCR10S_FltLte_1.zip)
I create my models in Sketchup and import the Sketchup STL files into Cura. Models imported into Cura 5.6.0 fail to slice.
This model slices without problem in Cura 5.4.0. I have tried several different STL models, including smaller and simpler models) and they have all failed to slice in Cura 5.6.0

closed 2024-04-03 by HellAholic

djardi, 2024-04-04:
> We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well?
> You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

I am happy to try this, but can you remind me of the name of my model? I do enough work with SketchUp and Cura that I do not remember which model had the problem.

David Jardine

HellAholic, 2024-04-04:
Hey @djardi , 
The model is attached as a zip to the github issue if you don't have it on your device. [CCR10S_FltLte_1.zip](https://github.com/Ultimaker/Cura/files/13808145/CCR10S_FltLte_1.zip)

djardi, 2024-04-05:
Thanks. I had forgotten which model had the problem. The zipped model That you emailed opened and sliced without any problems in the new version of Cura (5.7.0). Although I no longer had a copy of that file because I had done more work on the model, I tried slicing the newer version of the model. It also sliced flawlessly. Thanks so much for all the work on this and for the great Cura program. I am happy to be able to use the latest version of the program.

David Jardine