Ultimaker/Cura #17863: Can't slice fresh download

DBKobra, 2024-01-02:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Monoprice MP10 Mini (Custom)

### Describe your model

Brand new download. Default settings. Tried changing scaling and resolution to no avail.

### Add your .zip here ⬇️

[Fail.zip](https://github.com/Ultimaker/Cura/files/13808391/Fail.zip)

5axes, 2024-01-03:
An other "Print Thin Walls" issue : https://github.com/Ultimaker/Cura/discussions/16440
[image]

Duplicate : Slice Failed due to **Print Thin Walls**
https://github.com/Ultimaker/Cura/issues/17750
https://github.com/Ultimaker/Cura/issues/17725
https://github.com/Ultimaker/Cura/issues/17679
https://github.com/Ultimaker/Cura/issues/17676
https://github.com/Ultimaker/Cura/issues/17671
https://github.com/Ultimaker/Cura/issues/17606
https://github.com/Ultimaker/Cura/issues/17560
https://github.com/Ultimaker/Cura/issues/17528
https://github.com/Ultimaker/Cura/issues/17503
https://github.com/Ultimaker/Cura/issues/17499
https://github.com/Ultimaker/Cura/issues/17470
https://github.com/Ultimaker/Cura/issues/17467
https://github.com/Ultimaker/Cura/issues/17457
https://github.com/Ultimaker/Cura/issues/17456
https://github.com/Ultimaker/Cura/issues/17496
https://github.com/Ultimaker/Cura/issues/17478
https://github.com/Ultimaker/Cura/issues/17439
https://github.com/Ultimaker/Cura/issues/17438
https://github.com/Ultimaker/Cura/issues/17410
https://github.com/Ultimaker/Cura/issues/17367
https://github.com/Ultimaker/Cura/issues/17344
https://github.com/Ultimaker/Cura/issues/17177
https://github.com/Ultimaker/Cura/issues/16688
https://github.com/Ultimaker/Cura/issues/16515
https://github.com/Ultimaker/Cura/issues/16417

MariMakes, 2024-01-18:
Hi @DBKobra,

Quick update from our side 👋
We are introducing a new tool to analyze slicing crashes like the one you reported. 💪

Even with the changes we made in the latest versions of Cura we still see that your model crashes. 🫤 But we have added your project file to our analyzing tool and found there were at least 8 crashes that were similar to yours, which helps us to narrow down what the cause is of your crash so we can resolve this for you and others.
Hopefully, we can find a solution soon 🤞

I'll be closing this issue as a duplicate but you can follow the progress here https://github.com/Ultimaker/Cura/issues/16417

closed 2024-01-18 by MariMakes

HellAholic, 2024-02-16:
Small update,
Issue is resolved in CURA-11570 and will be part of the next Cura release.