Ultimaker/Cura #17868: Slicing failed on object sliced in previous version

RolfTO, 2024-01-02:
### Cura Version

5.6.0

### Operating System

Mac OSX Sonoma 14.2.1

### Printer

Ultimaker 2+

### Name abnormal settings

Used Taulman Bridge Nylon settings I created and used before

### Describe model location

Nope

### Describe your model

Plug in no longer on site

### Add your .zip here ⬇️

[Threaded_Shaft_Bigger_Knob.3mf.zip](https://github.com/Ultimaker/Cura/files/13810548/Threaded_Shaft_Bigger_Knob.3mf.zip)

RolfTO, 2024-01-02:
Sorry if not following proper protocol. n00b at github. First time Cura failed to slice. Object was merged in Meshmixer, so will try again there too.  I sliced and printed this object fine with a previous version of CURA, but may have used PLA as it was just an idea then. Again, sorry in advance if I made stupid mistake and am wasting your time.

HellAholic, 2024-01-02:
Hey @RolfTO 
Just took a quick look at your project file. 😖 Sorry to hear you are stuck.
I have a quick workaround for you:
- you can rotate the model slightly
- and then move it from the center of the build plate

that should resolve the Fail slice. I've also included a project file with the above changes implemented.
Hope that helps 👍 Let me know.

[UM2_Threaded_Shaft_Bigger_Knob.zip](https://github.com/Ultimaker/Cura/files/13810693/UM2_Threaded_Shaft_Bigger_Knob.zip)

RolfTO, 2024-01-02:
Thanks so much HellAholic. I actually spent the morning recreating the object in Fusion 360 because when I looked closely at the threads they were a mess of mesh. (I'm doing a favor for a fellow maker who provided that threaded shaft already in mesh and it was quite a garbage model. It's amazing it sliced at all.)  The new object sliced like a dream and is printing now.  Thanks again for looking at it. If I have trouble slicing in the future, I'll try the tricks above. Your help is much appreciated. Happy New Year!

HellAholic, 2024-01-03:
Glad to hear your problem is resolved :) I'll close this ticket since you're not experiencing the issue anymore.
Happy New Year!

closed 2024-01-03 by HellAholic