Ultimaker/Cura #17869: Not uninstalling old versions

Cyberdiver3750, 2024-01-02:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Monoprice

### Reproduction steps

Installed v5.6.0 

### Actual results

when I go to use it it still opens v5.4.0

### Expected results

v5.6.0 should have opened. Shouldn't thee old versions be uninstalled? If not then why isn'tv5.6.0 not associated with the .stl files as the default version?

### Add your .zip and screenshots here ⬇️

Thank you for the help

Asterchades, 2024-01-03:
By design, Cura doesn't remove the old versions. This isn't a default or anything that can be toggled - it just doesn't do it. Every release version should be able to happily coexist beside any others so this shouldn't cause any problems, although minor revisions will likely share their settings (eg 5.3.1 will share with 5.3.0). I'm afraid I'm not privy to the reasoning behind this, and can only assume it's to aid in evaluation so new features and versions can be tested without interrupting existing workflows. Ultimaker does market their printers to professionals, after all, so the accompanying software has to be designed with that market mind.

In terms of file association, the installer has not made any alterations here since Cura 5 launched - only Cura 4 and earlier would provide the option to set file associations (and I'm not even entirely sure these work under Windows 8 or newer). So any existing file association will have been done manually, and will subsequently be maintained following the installation of a newer version. You will need to update these yourself using the Windows Explorer context menu (right click) on an STL. NB You may need to hold down [Shift] to reveal the "Open with..." option.

Cyberdiver3750, 2024-01-03:
Thank you very much for the reply. It was very helpful.