Ultimaker/Cura #17875: Failed to slice

Gurba667, 2024-01-03:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Creality CR-10S Pro V2 with Sonic Pad/Moonraker

### Name abnormal settings

Tree support seems to be the issue. Works fine with normal support.

### Describe model location

Have tried various directions with same result.

### Describe your model

Have sliced before with a different side down on build plate.

### Add your .zip here ⬇️

[Skuffbrakett.zip](https://github.com/Ultimaker/Cura/files/13816467/Skuffbrakett.zip)

Vandresc, 2024-01-19:
👋 Quick update on our side.
Your issue seems to have been resolved with this fix from our wonderful community member ThomasRahm/
(https://github.com/Ultimaker/CuraEngine/pull/2002)
 

I'll close this issue since it's resolved, a fix should be available with the upcoming release. 

Thanks again, and please let us know if you run into any other issues 💪

MariMakes, 2024-01-19:
Even better news 😊, this fix is also included in the 5.7 Alpha, which has been released to analyze the incoming slicing crashes.
If this crash is blocking you in any way you can download the nightly build to continue printing (and to help us analyze the slicing crashes) 

You can find the downloads here https://github.com/Ultimaker/Cura/discussions/18080