Ultimaker/Cura #17887: [5.6.0] Cura generates filled layers not present in the original 3MF file

georgmierau, 2024-01-03:
### Cura Version

5.5.0, 5.6.0

### Operating System

Windows 11

### Printer

Elegoo Neptune 3 Pro

### Reproduction steps

Add [this model](https://drive.google.com/file/d/1CVXZtxBBE7GIb0IyZdIE_e4yHWzVcHjB/view?usp=sharing) and slice it using 0.12 mm layers.

### Actual results

In this orientation it generates filled layers not present in the original model. There are no additional layers generated if I'm using an STL file instead of 3MF or if the 3MF model is rotated 90° around the z-axis. This issue is also not present in PrusaSlicer 2.7.1 (same 0.12mm layer height).

### Expected results

No additional filled layers not present in the 3MF model.

### Add your .zip and screenshots here ⬇️

Demo: https://imgur.com/a/aPQYlix

[EN3PRO_gunpla_organizer.zip](https://github.com/Ultimaker/Cura/files/13823822/EN3PRO_gunpla_organizer.zip)

GregValiant, 2024-01-05:
Thanks for the report.
That's a bad model.  It has some flipped surfaces in it and it isn't watertight.  I tried to repair it in MS 3D Builder but it still had problems.  If you go to the Cura Marketplace and load "Mesh Tools" then when you load a model you will get a pop up message if errors are found in the model.
[image]

This report is from "formware.co/onlinestlrepair"

-> Analysed your file:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> 2 Non-manifold edges (?)
--> 4 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%

So there aren't a lot of problems but they are making the slice incorrect.  This is an interesting view.  The original model is on the right.  The left "repaired" model has been spun 180° so the same area is in the image.
You can see that the extrusion lines came out multi-colored like Cura couldn't decide if they should be walls, or skins, or what they should be.  This is an excellent example of what can go wrong when the model isn't right.
[image]

I'll go ahead and remove the bug label and close this.  I hope you understand.

georgmierau, 2024-01-05:
I don't really get your arguments since rotating the "not watertight" model 90° around z-axis in Cura (!) makes it "watertight" for some reason.

GregValiant, 2024-01-05:
It might have something to with the way the slice plane is hitting the open seam.  That model slices fine when it's fixed and Cura gives a warning that the model isn't right.  If the seam is in a position where the slicer wraps around and enters the interior of the model then it can't tell where the outside ended and where the inside started.  So rotating the model might be hiding the error (there are only 2 open seams).  

I went back and used different Cura versions.  4.13.1 noted that the model has errors and then sliced it correctly, none of the 5.x versions were happy with this model.  I'll put the bug label back on this.  As an aside, PrusaSlicer didn't have any trouble with it either.

georgmierau, 2024-04-03:
>  I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? 

The same problem persists:

https://imgur.com/a/aPQYlix

Slicing result (the presence of the additional layers) depends on the orientation of the model. So, no, nothing was "solved" here.

HellAholic, 2024-04-03:
- You are using a custom profile with a custom layer height but none of the other settings have been adjusted to compensate for the change to the layer height. Selecting either of the included profiles for 0.1mm or 0.15mm layer height resolves the issue. 
- Changing the "wall line count" to either 2 or 4 in your profile resolves the issue for all orientations.

I'm sure there are other settings that you can adjust to get the desired results without modifying the layer height from 0.12mm or changing the wall line count number.

If you notice this issue outside of the discussed conditions you can notify us. 

The reason you received the half-automated notification for issue has been resolved is the slice crash tag. I'll take that one out.

HellAholic, 2024-04-03:
Ticket added to the board.
Internal reference CURA-11805