Ultimaker/Cura #17889: Neptune 4 Plus | Saving gcode error (not responding)

tsweeney33, 2024-01-03:
### Cura Version

4.8.0-231103.0

### Operating System

Windows 11

### Printer

Neptune 4 Plus

### Reproduction steps

[image]

### Actual results

[image]

### Expected results

The file should have downloaded to my pc, but instead it has a "not responding error".

### Add your .zip and screenshots here ⬇️

none

GregValiant, 2024-01-05:
Thanks for the report.
4.8 is about 15 versions old.  Is that the Ultimaker version or the Elegoo version (that might have come with the printer)?

tsweeney33, 2024-01-05:
> Thanks for the report.
> 4.8 is about 15 versions old. Is that the Ultimaker version or the Elegoo version (that might have come with the printer)?

Just bought the printer and downloaded software from flash drive I got. I also uninstalled and downloaded the latest version for the Neptune 4 Plus on Elegoos website. Still not working.

GregValiant, 2024-01-07:
Go [HERE](https://github.com/Ultimaker/Cura/releases?page=3) and scroll down to version 4.13.1.  It's older but very stable.  Click on the "Assets" link and a list of the installers will come up.  Use the AMD64 link for your Windows setup.
You don't have to un-install the older versions.  Creality, Elegoo, and other printer manufacturers put out their own versions of Cura.  They aren't exactly the same as UM Cura.
There have been issues between the newer Cura versions and Windows 11.  4.13.1 works really well but I'm running Win10 so I don't know if there are compatibility issues with Win11 or not.