Ultimaker/Cura #17890: Way to get initial unused extruder number with arithmetic as a printer definition setting

ansonl, 2024-01-04:
### Is your feature request related to a problem?

When initially priming a nozzle there currently is no way to find the opposite of the `initial_extruder_nr`. With a dual nozzle printer, it would be useful to prime the nozzle that not the `initial_extruder_nr` first and then prime the `initial_extruder_nr` nozzle. This way the last primed nozzle immediately started printing after priming rather than guessing which nozzle will be used first in the machine start gcode. 

### Describe the solution you'd like

Either exposing the `initial_extruder_nr` or `getExtruderManager().getInitialExtruderNr()` as a value that can be referenced by settings in `fdmprinter.def.json`. 
OR
Implementing Gcode math that is interpreted during the slicing process similar to PrusaSlicer/BambuStudio.

I tried to get the unused extruder by adding the below setting to fdmprinter.def.json but I cannot seem to access the `initial_extruder_nr` or `getExtruderManager().getInitialExtruderNr()` values from a setting since `initial_extruder_nr` is set during the slicing process.

```
"initial_first_unused_extruder_nr":
                {
                    "label": "Initial First Unused Extruder",
                    "description": "The first unused extruder number.",
                    "unit": "",
                    "type": "int",
                    "resolve": "getExtruderManager().getInitialExtruderNr()",
                    "default_value": 0,
                    "value": "resolveOrValue(1-'initial_extruder_nr')",
                    "minimum_value": "0",
                    "minimum_value_warning": "max(extruders_enabled_count)",
                    "maximum_value": "16",
                    "enabled": true,
                    "settable_per_mesh": false,
                    "settable_per_extruder": false,
                    "settable_per_meshgroup": false
                },
```

### Describe alternatives you've considered

Currently dual color prints that I've done have started printing with the first extruder. But this is not guaranteed.
I prime the second extruder first, then prime the first extruder. Then I set both extruders to the standby temperature and immediately raise the `initial_extruder_nr` extruder to the print temp before priming it again and starting the print. 

The machine start gcode does not know which extruder will be the initial one so priming three times ensures that both nozzles are primed and the `initial_extruder_nr` extruder is the last primed nozzle. 

### Affected users and/or printers

All users with dual nozzle printers.

5axes, 2024-01-04:
Just a suggestion: why not use the order ? : 

{infill_extruder_nr} or {wall_extruder_nr}
{adhesion_extruder_nr}

ansonl, 2024-01-04:
@5axes I haven't set an infill or wall extruder because the dual color object uses each extruder for its own object's infill and walls. When no extruder overrides all infill or wall, the number set is -1. 

Ex: 
[image]

I tried `; initial_first_unused_extruder_nr = {extruders_enabled_count - (initial_extruder_nr + 1)}` but math is not allowed in the gcode, it slices and exports as the same thing below which is why I tried the settings way as well. 

```
; initial_first_unused_extruder_nr = {extruders_enabled_count - (initial_extruder_nr + 1)}
```

5axes, 2024-01-04:
> I tried `; initial_first_unused_extruder_nr = {extruders_enabled_count - (initial_extruder_nr + 1)}` but math is not allowed in the gcode, it slices and exports as the same thing below which is why I tried the settings way as well

If you use the last release 5.6 you can use this code